RobotLocomotion/drake #1608: installation issue on windows

XuanLin, 2016-01-14:
Hi everyone,

I installed drake on my windows 10 system. At the end of the installation I am getting this error:

"C:\Users\Xuan\Documents\simulation\drake-distro\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\Xuan\Documents\simulation\drake-distro\pod-build\drake.vcxproj" (default target) (10) ->
(CustomBuild target) ->
  CUSTOMBUILD : error : MEXEXT: Unsupported platform. [C:\Users\Xuan\Documents\simulation\drake-distro\pod-build\drake.
vcxproj]

I run the cartpole example and it is working, but when I try Hongkai's optimal grasp :
testIKGrasp.m

I get this error:

Error using checkDependency (line 442)
Cannot find required dependency: spotless

Error in testIKGrasp (line 5)
checkDependency('spotless');

and we figure out that spotless is downloaded but not installed, so I manually installed it (run spot_install.m), then another error occurs:

Error using checkDependency (line 442)
Cannot find required dependency: lcmgl

Error in testIKGrasp (line 10)
checkDependency('lcmgl');

So we guess that there are still some part of drake which is not installed. Does anyone know how to fix it?

I follow the instructions on wiki and Cmake/Gnu are here:
[image]

I am using Microsoft Visual Studio 2015 Community as compiler, and the system variable looks like this:

[image]

[image]
(it looks like there is a mismatch on Cmake)

Thank you!

hongkai-dai, 2016-01-19:
Is this still a bug? Which version of cmake are you using? According to this page, http://answers.opencv.org/question/44589/opencv-30-problems-building-dlls-and-libs/, you might check your cmake version, and the linker setting. @RussTedrake , have we tested drake installation on a Windows 10 platform?

RussTedrake, 2016-01-19:
I've not tested on Windows 10 recently -- i used to, but then someone told me that very few people use windows 10.  :)

If there are many users out there, it would be good to know it.

XuanLin, 2016-01-20:
I get Cmake from the link given by wikipedia, so that should be the latest version-3.4.2. I can double check it tomorrow when I reach out to my laptop.

By the way it is true that more people work on linux in our lab. But our group do need windows for some specific reason.

RussTedrake, 2016-01-20:
We definitely want to support windows.  But why windows 10?  I thought most people had decided that Windows 7 was still the best for now…?

XuanLin, 2016-01-20:
Yes, people update to win 10 and quit due to issue of driver support, I just haven't met any issue yet. I can go back if it's needed.

RussTedrake, 2016-01-20:
I’m sure it can work.  Just trying to gage the audience.

XuanLin, 2016-01-21:
I guess when we are not actually communicating with hardware then windows 10 is fine. People in my lab tend to put win10 on personal computer and doing hardware control on a linux machine (I do it on the lab computer, which doesn't have matlab). 
Anyway, let me know if you are going to test it or not, so I can make decisions.
Thanks!

RussTedrake, 2016-01-21:
We’ll test it, but probably not in the next few weeks.

RussTedrake, 2016-01-23:
what happens when you run `mex -setup` at the matlab prompt?

XuanLin, 2016-01-23:
[image]

RussTedrake, 2016-01-24:
Is it also set up for c++?

XuanLin, 2016-01-24:
I didn't do mex -setup C++, you want me to try if this solves the issue?

RussTedrake, 2016-01-24:
Yes please.  That's relatively new in matlab that they're separate.  Perhaps we need to make the install instructions clearer..?

XuanLin, 2016-01-24:
The issue is still there...
[image]

One question:  When I type make option and open up a Cmake, what should I put in "Where is the source code"? I just put "C:/Users/Xuan/Documents/GitHub/drake-distro", so it looks like this:
[image]

XuanLin, 2016-01-24:
I double try it. The error occurs right here, if it's useful:
[image]

RussTedrake, 2016-01-24:
Is your matlab installed on the P drive? 

Try this:  edit drake-distro/drake/cmake/mex.cmake, line 111.

change 
```
execute_process(COMMAND ${mexext} OUTPUT_VARIABLE MEX_EXT OUTPUT_STRIP_TRAILING_WHITESPACE)
```
to
```
execute_process(COMMAND "${mexext}" OUTPUT_VARIABLE MEX_EXT OUTPUT_STRIP_TRAILING_WHITESPACE)
```
and let me know.

RussTedrake, 2016-01-24:
you might have to do something similar on line 124.  `${mex}` to `"${mex}"`

XuanLin, 2016-01-25:
Do you mean this:
[image]

I am not sure... When I install drake, I do mex -setup -> git clone -> make options -> make -> addpath_drakeup, up to stetup make options the cmake folder is still empty, so do you mean I that I edit after installation? But the error occurs during installation...

RussTedrake, 2016-01-25:
That's correct.  Take the current partially-installed directory and make these local edits.  The run make again to see if it helps

XuanLin, 2016-01-27:
Unfortunately no, the error persists, although it takes less time to finish.
[image]

RussTedrake, 2016-01-27:
This is hard to debug remotely.  For what it’s worth, it looks the same as the issue here:
http://answers.opencv.org/question/44589/opencv-30-problems-building-dlls-and-libs/

Can you confirm that your matlab is installed in the P:\ drive?  (perhaps send the complete path) 
Do you have multiple versions of matlab installed on the machine?  (is there any chance you’re running mex -setup in one, and then cmake is finding a different one)?

XuanLin, 2016-01-29:
Sorry for the delay here...

Matlab is on P drive, with directory:
P:\Program Files\MATLAB\Matlab2015b\bin\Matlab.exe

I don't have multiple Matlabs on my laptop. Anyway, I will find some way around, maybe use a different operating system.

Thank you!

closed 2016-02-27 by RussTedrake