RobotLocomotion/drake #1621: Independent LCM types pod?

mauricefallon, 2016-01-19:
As folks on the DRC project know we have a large set of LCM types. Some of these are general purpose: joint_states_t, joint_commands_t as well as message types in libbot.

ROS defines a number of messages types in independent groups e.g. sensor_msgs, geometry_msgs. This eases interaction between systems.

@patmarion and I discussed perhaps making an independent library of lcm types - or perhaps expanding libbot's types. Would folks developing drake use such a dependency or see worth in doing this?

RussTedrake, 2016-01-19:
Yes, I like that idea.  There are a number of types that I’ve been about ready to duplicate, and would prefer not to (but don’t want to encur all of openhumanoids as a dependency).

mauricefallon, 2016-01-19:
What do you think about expanding libbot lcmtypes?
Obviously, an issue with doing this is that changes to public types effect all users.

RussTedrake, 2016-01-19:
I don’t consider libbot a required dependency anymore; in fact I consider it scheduled for deletion (we’re the only ones updating it, and it’s getting worse to compile as compilers move forward).  I don’t have any problem with those types, but think we should separate the types from libbot.

jhoare, 2016-03-23:
Was this solved by bringing the bot2_lcm_datatypes in? (#1820) Or is this something else beyond that?

RussTedrake, 2016-03-23:
yes.  resolved.

closed 2016-03-23 by RussTedrake