RobotLocomotion/drake #1625: valkyrie texture maps show in open-humanoids but not in drake-visualizer

RussTedrake, 2016-01-19:
this is also blocking #1583 -- @sproulx cannot get his texture maps to display in drake-visualizer.

@patmarion -- can you tell us what texture maps are supported?  or point us to the code?  is there a code snippet from open-humanoids that could be pulled over to drake-visualizer?

patmarion, 2016-01-20:
I can repeat the error on my computer.  It appears that textures have never worked for the Valkyrie urdf in drake-visualizer.  The reason is that the textured mesh files were generated with a different file extension than expected by drake-visualizer.  I could generate the meshes in the other style, but I prefer to extend drake-visualizer to make it support the alternative extension.  I've put together this feature on my laptop, so this issue will be resolved by an incoming pull request to bump forward the director sha1.

RussTedrake, 2016-01-20:
But why did it work in the other director app?

patmarion, 2016-01-20:
In director, the urdf visualization is managed by a c++ class called ddDrakeModel which uses the drake c++ api to parse urdf.  In drake-visualizer, the mesh loading is handled by a python class which is listening to the DRAKE_VIEWER_LOAD_ROBOT channel.  Basically, two separate implementations.

RussTedrake, 2016-01-20:
That’s a bit ironic, that the drake-visualizer app uses less of drake’s API.  :)   Seems to me like converging those two might make sense, but I’ll of course defer to you.  Thanks Pat!

patmarion, 2016-01-20:
Haha yeah, it's true.  But, drake-visualizer doesn't (currently) need to know about urdf or kinematic trees.  The two applications share common code to load meshes (using vtk), but director gets filenames from RigidBodyTree, while drake-visualizer gets filenames from messages on the DRAKE_VIEWER_LOAD_ROBOT channel.

RussTedrake, 2016-01-20:
That makes sense.  Note that director is now the only client supported by BotVisualizer, so it’s also fine if we need to add logic on the drake side to support it (e.g. file name wrangling)

patmarion, 2016-01-21:
I think this issue can be closed now.

closed 2016-01-21 by RussTedrake