RobotLocomotion/drake #1627: need snopt alternative for c++ nonlinear optimization

RussTedrake, 2016-01-19:
I think NLopt is probably the right choice.  Here is a note on this from @hongkai-dai:

I did a quick search, and it seems IPOPT is a good choice. Both IPOPT and and NLopt are free and open source. IPOPT works on Cygwin or MinGW http://www.coin-or.org/Ipopt/documentation/node15.html, while NLopt provides a binary file for windows. http://ab-initio.mit.edu/wiki/index.php/NLopt_on_Windows

NLopt supports many algorithms, but the one that can handle nonlinear constraints is SLSQP (sequential least squares quadratic programming) http://ab-initio.mit.edu/wiki/index.php/NLopt_Algorithms#Local_gradient-based_optimization. Unfortunately it uses dense matrix to compute the Hessian, so not a good fit for problem with several thousand variables. On the other hand, IPOPT supports sparse gradient.

In terms of computation speed, since NLopt is relatively young (initially released in 2008), I found only one paper on the comparison. https://www.jstage.jst.go.jp/article/jrsj/32/6/32_32_536/_pdf. They compare NLopt and IPOPT on some small size problems (no more than 5 decision variables), generally NLopt is faster than IPOPT (mostly twice as fast).

bradking, 2016-01-28:
For reference, there is also Sandia's [OPT++](https://software.sandia.gov/opt++/) under LGPL (Subversion repo [here](https://software.sandia.gov/svn/public/optpp/optpp/trunk/)), but I don't know whether it meets drake's requirements.

RussTedrake, 2016-01-28:
I hadn’t actually looked at that one.  One of the algorithms it provides is a wrapper to NPSOL (written by the same folks as SNOPT).  The fact that it optionally takes second derivatives is great.  But it looks like it does not use sparsity information, so probably won’t perform as well in our big trajectory optimization uses.

I’d be happy to include it on the wishlist. 

- Russ

tkoolen, 2016-02-01:
There's also Eigen's unsupported NonLinearOptimization module, which is a port of MinPack http://eigen.tuxfamily.org/dox/unsupported/group__NonLinearOptimization__Module.html

RussTedrake, 2016-02-01:
Wow.  I didn’t realize.  That looks extremely relevant and easy to try.

tkoolen, 2016-02-01:
Browsed the source code a bit; looks like it's only unconstrained optimization.

RussTedrake, 2016-02-01:
Ah.  Too bad.

Lucy-tri, 2016-03-07:
I downloaded NLOpt and started looking at it. @RussTedrake, did you want to assign this to me?

Lucy-tri, 2016-03-18:
Status update:  I've been working on integrating NLOpt into Drake.  NLOpt is now building within Drake (on our fork). I'm now working on writing the wrapper. I'm getting help from others.

sammy-tri, 2016-03-24:
I've been assisting with integrating NLopt into Drake as an alternative solver to SNOPT,  with some amount of success (though my development branch is going to need a significant cleanup, but that's another story...  when #1874 lands I can start cleaning it up)

I've gotten most of the tests in testOptimizationProblem to pass (using SLSQP algorithm), including another I added (quadratic objective, linear + bbox constraints, nothing particularly new) except for the trivialLeastSquares test, which consists entirely of LinearEqualityConstraints (and eventually one BoundingBoxConstraint with bounds wide enough that it has no effect on the solution).

SNOPT handles LinearEqualityConstraints through a different mechanism than other constraints (iAfun, jAvar, A arguments), so I don't have much to work with as inspiration there.

Using NLopt's add_equality_mconstraint doesn't seem to do the right thing at all on this problem (which I suspect is unsurprising, as SLSQP is looking for nonlinear inequalities?  I'm a little out of my element).  Treating the LinearEqualityConstraint as a generic constraint doesn't work in SNOPT either, but that might be for different reasons.

Any suggestions from someone who understands the underlying algorithms a bit better?  I'll add a link to the code once I've got cleanup a bit further underway.

hongkai-dai, 2016-03-25:
I have not looked into the code yet, but NLopt says the SLSQP algorithm supports nonlinear equality constraints http://ab-initio.mit.edu/wiki/index.php/NLopt_Algorithms#Local_gradient-based_optimization, so could you treat the linear equality constraints as nonlinear equality constraints in NLopt, as a constraint with value `f(x) = A*x+b`, and gradient being `A`?

sammy-tri, 2016-03-25:
I've done a minor cleanup/rebase of my working state, which is now at https://github.com/sammy-tri/drake/tree/nlopt_integration -- it's a bit of a mess in there right now, and rather heavy on the printf debugging.

The approach discussed above is effectively what I'm doing now, and NLopt doesn't seem happy with it.  Specifically, it throws nlopt::roundoff_limited at the end of the first iteration after evaluating the constraints once.  (this is with the bounding box constraint disabled, since that causes all new problems (it also doesn't affect the outcome since it's a wide constraint and snopt is fine either way)).

I've dug around in the debugger trying to see what's going on inside of SLSQP  (nlopt/slsqp/slsqp.c).  The slsqp() function returns with mode == 6, which caused the roundoff_limited error.  Mode 6 is documented in a couple of ways, as "singular matrix C in LSQ subproblem" or "MATRIX C IS NOT OF FULL RANK".  The mode bit is actually set in the function lsq_() on line 1135...   and at that point I'm trying to back out what happened from code which looks like machine-translated FORTRAN without really understanding the math to begin with.

I feel like I might have been providing too much detail in the wrong direction there...

sammy-tri, 2016-03-28:
Hmm...   It looks like my issues may be entirely because I'm calling nlopt::add_equality_mconstraint() in order to try to speed things up by evaluating multiple constraints in the same functions.  I looked at the test a bit more, and we only have LinearEqualityConstraints which have multiple outputs in testOptimizationProblem, there are no generic or linear constraints which test this case.

(see http://ab-initio.mit.edu/wiki/index.php/NLopt_C-plus-plus_Reference#Nonlinear_constraints for a discussion of how NLopt supports scalar or vector constraints).

I've got something working in my tree which doesn't use the mconstraint() version with more than one output, and it does pass all tests.  I'm going to move in the direction of getting my tree cleaned up enough to work using only the NLopt scalar constraint form and we can investigate the performance issues later.

Lucy-tri, 2016-03-28:
Wow, that sounds terrific. Sounds like you got it working.
Lucy

sammy-tri, 2016-04-07:
Now that #1962 has been merged, should we close this issue?  My thought is yes (though noting that #2019 is still outstanding)

closed 2016-04-07 by RussTedrake