RobotLocomotion/drake #1630: Reconfigure causes unnecessary rebuild

patmarion, 2016-01-19:
If I run make from drake/pod-build with all targets up to date, it runs very quickly, nothing rebuilds.  If I reconfigure cmake, then make quick recompile some trivial files and re-link almost everything, this takes a long time.  This often occurs when cmake configure rewrites generated files which are compile dependencies.  Sometimes this is easy to fix, sometimes you have to implement more robust dependency logic.  In this case, maybe it's from Path.h?  There might be others.  Path.h could be generated with the configure_file() command which properly tracks dependencies, rather than file(WRITE).

RussTedrake, 2016-01-19:
I'll bet you're right.  Path.h is probably causing it.  Good call.

patmarion, 2016-01-20:
If you'd like, I can try writing a pull request to use configure_file() to generate Path.h.  I would make a new file called Path.h.in in the source director which cmake will copy to the build generated directory with the configure_file() command.

RussTedrake, 2016-01-20:
Sounds great.  Thank you.

commit 19fe982c (2016-01-20): use the configure_file() command to generate Path.h.  Fixes #1630

This avoids touching the file on successive reconfigures. The file will
only be touched if the cmake variables it depends on are modified.

Note, removes usage of ADDITIONAL_MAKE_CLEAN_FILES.
commit a0a87907 (2016-01-20): Merge pull request #1632 from patmarion/configure-path-header

use the configure_file() command to generate Path.h.  Fixes #1630
closed 2016-01-20 by RussTedrake