RobotLocomotion/drake #1639: Query Optimization Program Class

mposa, 2016-01-21:
Feature request from our discussion today: a method to query the type of an optimization program (LP,QP, etc). 

Optionally, also to define an optimization program in a way such that the type is immutable. For example, define as an LP and throw an error for any nonlinear costs or constraints.

jwnimmer-tri, 2019-03-26:
This is available in rough form now with the "attributes" on a program.  The request here is for sugar to convert attributes into QP/LP/etc.

jwnimmer-tri, 2021-11-11:
We how have [GetProgramType](https://github.com/RobotLocomotion/drake/blob/master/solvers/get_program_type.h) available.

We don't yet have a way to configure a MathematicalProgram to pre-suppose a nominal type and then throw in case the user exceeds that type when adding costs or constraints.  As a starting point, the user can check the MP type after building it, and then throw their own exception.  If we find that too onerous, we can try to work on the sugar for it (but let's open a new issue in that case).

closed 2021-11-11 by jwnimmer-tri