RobotLocomotion/drake #1641: Drake parallel build support

patmarion, 2016-01-21:
Sorry to drag this issue up again, but Drake now has enough C++ build targets that make -j (unlimited jobs) thrashes on my laptop.  However, make -j4 results in a serial build because of the usage of "cmake --build".  I recommend we find a way to use "$(MAKE)" where appropriate.  I believe it would have to be updated in both top level CMakeLists.txt and drake/Makefile.  I'm happy to write a PR for this if it sounds ok to you.  I know at one point it was using $(MAKE) but I forget why that mode lost support?

RussTedrake, 2016-01-22:
But we do use $(MAKE) when it works, I think??  Line 193 of drake-distro/CMakeLists.txt.

patmarion, 2016-01-22:
I think PODS_MAKE_COMMAND is set to $(MAKE) and is used for externals, but when calling ExternalProject_Add for drake, on line 330:

<pre>
BUILD_COMMAND ${CMAKE_COMMAND} --build ${PROJECT_SOURCE_DIR}/drake/pod-build --config $<CONFIGURATION> --target install
</pre>

RussTedrake, 2016-01-22:
Ah.  Also for the CMAKE externals.  Is $(MAKE) really preferrable to ${CMAKE_COMMAND} ?  I think the logic around PODS_MAKE_COMMAND was the way we addressed the previous concerns.  I’m ok updating the CMAKE external / drake logic if that is the recommended soln.

billhoffman, 2016-02-04:
$(MAKE) is good when the generator is set to Unix Makefiles, any PR would have to include a check for the make generator.

mwoehlke-kitware, 2016-02-17:
> $(MAKE) is good when the generator is set to Unix Makefiles, any PR would have to include a check for the make generator.

See https://github.com/Kitware/fletch/blob/master/CMakeLists.txt (look for MAKE_EXECUTABLE) for a good example of this...

mwoehlke-kitware, 2016-02-17:
Actually, why is it necessary to override the build command at all? EP_Add will DTRT if not overridden.

billhoffman, 2016-02-18:
@mwoehlke-kitware I would say experiment and report back.  It should be pretty easy to tell if it is actually doing parallel builds or not.

billhoffman, 2016-02-26:
@mwoehlke-kitware we can close this right?

mwoehlke-kitware, 2016-02-26:
Yes; it should have been closed when #1729 was merged...

closed 2016-02-26 by RussTedrake