RobotLocomotion/drake #1643: NonlinearProgram: MajorStepLimit

taylorott, 2016-01-21:
In the snopt user guide, it says that there is an option, Major Step Limit, to set the major step size limit. This seems to be one of the snopt options that is not present within the NonlinearProgram class. I tried to hack it by introducing a line within the NonlinearProgram class:

snsetr('Major Step Limit',.1)

However, this did not  work. Is there currently a way to set this option? Alternatively, is there a different option I can set to bound the step size used by the solver?

hongkai-dai, 2016-01-21:
Could you check the SNOPT you are using? If you type
```
checkDependency('NonlinearProgramSnoptmex')
```
and it returns 1, then `NonlinearProgram` is not calling SNOPT directly; instead it is calling our SNOPT wrapper function `NonlinearProgramSnoptmex`. You will need to modify `drake/solvers/NonlinearProgramSnoptmex.cpp`. You can add your snopt options after line 206 of `NonlinearProgramSnoptmex.cpp`, and also add that option in function `setSolverOptions` in `NonlinearProgram.m`

taylorott, 2016-01-21:
It returned 1. I'll try making the modifications. Thanks!

closed 2016-01-21 by RussTedrake