RobotLocomotion/drake #1647: Test new version of SNOPT

hongkai-dai, 2016-01-23:
The SNOPT version we are using now is 7.2. The newest version on http://www.sbsi-sol-optimize.com/asp/sol_product_snopt.htm is SNOPT 7.5. The interface is changed slightly between the two versions. In the newer version, it requires to pass the multiplier (xmul,Fmul,xstate,Fstate) as the input to the snopt call, in the MATLAB interface. But the C interface look the same to me. @javieralonso get the C++ interface of SNOPT 7.5 working on his machine.

jwnimmer-tri, 2017-06-19:
See also #4611.

jwnimmer-tri, 2018-01-31:
I'm taking this, because it relates to #7240.

jwnimmer-tri, 2018-02-07:
We need some plan to obtain the SNOPT 7.5 source code for Drake use.  TRI has our own source copy and license, but for the Drake CI builds we are defaulting to the RL/snopt private repository and its academic license.

Proposal: obtain the academic-licensed source tarball, unzip it, and push to the RL/snopt repository unmodified as a tag (not master).  I think this would be enough for Drake's CI build and Drake developers to use.

Does that sounds good @jamiesnape and @RussTedrake?

\CC @EricCousineau-TRI

RussTedrake, 2018-02-07:
That’s a private repo so should be fine.  But why not just update RL/snopt/master?

jwnimmer-tri, 2018-02-07:
The current master has CMakeLists files and other various RL customizations that have accrued over the years.  I just want a pristine upstream source image, and assumed that removing all of those customizations was likely to disturb someone.  Conceivably we could _merge_ the 7.5 sources into master at some point, but I'd still want a pristine upstream tag somewhere (since it will mimic what users who download the sources themselves, without RL GH bits, would see).

RussTedrake, 2018-02-07:
aok with me.

jamiesnape, 2018-02-07:
Fine by me, though CI could happily download a source archive from somewhere private to take Git out of the loop.

jwnimmer-tri, 2018-02-13:
My is is first to gets 7.5 correctness using the git approach #8060) to get the upgrade done expediently, and then follow-up with source tarball coverage for #7240 in a second pass.

jwnimmer-tri, 2018-03-22:
The upgrade was reverted in #8348.  We need to test and debug why asan was unhappy.

jwnimmer-tri, 2018-03-22:
Calling `prog.SetSolverOption(SnoptSolver::id(), "Print file", print_file);` is a causal factor.  It looks to me so far like scratch strings in global rw data that are used during output formatting lack trailing nil bytes, which used to be overlooked probably because the data after them was already zero, but now for some reason asan is finding.

PR #8263 (2018-03-13): Switch to SNOPT 7.6.1 2018-01-20 release
Closes #1647.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8263)
closed 2018-05-11 by jwnimmer-tri