RobotLocomotion/drake #1649: Master doesn't compile on g++ 4.8

tkoolen, 2016-01-24:
On 4.8.5:
```
Linking CXX executable ../../bin/testVector
CMakeFiles/testVector.dir/testVector.cpp.o: In function `Drake::getDrakePath()':
testVector.cpp:(.text+0x6b): undefined reference to `__cxa_guard_acquire'
testVector.cpp:(.text+0x90): undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'
testVector.cpp:(.text+0x9a): undefined reference to `__cxa_guard_release'
testVector.cpp:(.text+0xa9): undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
testVector.cpp:(.text+0xc6): undefined reference to `__cxa_guard_abort'
```
4.7 and 4.9 are fine.

rdeits, 2016-01-26:
g++ 4.8 is also broken w.r.t. std::regex, see: https://github.com/RobotLocomotion/drake/issues/1650#issuecomment-175198691

RussTedrake, 2016-02-17:
now that #1663 is resolved, let's update the minimum gcc version to 4.9.

jamiesnape, 2016-02-17:
@RussTedrake Would you like the Jenkins builds to be updated to use GCC 4.9?

RussTedrake, 2016-02-18:
what version is jenkins using?  do we understand why it is building successfully with a more minimal version, while others have had trouble?  I would much prefer to not increment the minimal version if we can avoid it.

jamiesnape, 2016-02-18:
GCC 4.8 or Clang 3.7. Just responding to your comment above about updating the minimum version. My personal recommendation would be to support GCC 4.8 as it is so commonly used.

tkoolen, 2016-02-18:
The issue occurred while compiling a mex function, which Jenkins doesn't do yet.

billhoffman, 2016-03-16:
@RussTedrake can this issue be closed now that 4.9 is the minimum compiler?

jamiesnape, 2016-03-16:
It will close automatically once #1862 merges.

PR #1862 (2016-03-24): Ensure Clang/GCC are at least the minimum version
This closes #960, closes #1281, and closes #1649.

PTAL @RussTedrake.
closed 2016-03-24 by RussTedrake