RobotLocomotion/drake #1650: Building master with g++ 4.7 results in sleep_until compile error

tkoolen, 2016-01-24:
This is with 4.7.3 from apt-get on Ubuntu 14.04.

Error:
```
In file included from /home/twan/code/drake-distro/drake/../drake/systems/LCMSystem.h:11:0,
                 from /home/twan/code/drake-distro/drake/systems/LCMSystem.cpp:3:
/home/twan/code/drake-distro/drake/../drake/systems/Simulation.h: In function ‘void Drake::handle_realtime_factor(const TimePoint&, double, double, double)’:
/home/twan/code/drake-distro/drake/../drake/systems/Simulation.h:41:9: error: ‘sleep_until’ is not a member of ‘std::this_thread’
make[3]: *** [systems/CMakeFiles/drakeLCMSystem.dir/LCMSystem.cpp.o] Error 1
make[2]: *** [systems/CMakeFiles/drakeLCMSystem.dir/all] Error 2
In file included from /home/twan/code/drake-distro/drake/../drake/systems/LCMSystem.h:11:0,
                 from /home/twan/code/drake-distro/drake/../drake/examples/Pendulum/Pendulum.h:6,
                 from /home/twan/code/drake-distro/drake/core/test/testVector.cpp:2:
/home/twan/code/drake-distro/drake/../drake/systems/Simulation.h: In function ‘void Drake::handle_realtime_factor(const TimePoint&, double, double, double)’:
/home/twan/code/drake-distro/drake/../drake/systems/Simulation.h:41:9: error: ‘sleep_until’ is not a member of ‘std::this_thread’

```

Possibly related:
https://github.com/Homebrew/homebrew-versions/issues/262

RussTedrake, 2016-01-24:
Wow.  i had no idea that gcc was so flakey between versions.  that’s pretty ridiculous, right?

tkoolen, 2016-01-24:
I don't understand this one, since drake006 uses 4.7 and has compiled master just fine, correct?

RussTedrake, 2016-01-24:
Perhaps different minor releases of 4.7?  it looks like many are ok…?

tkoolen, 2016-01-25:
Both the version I installed on my machine and the version on drake006 are 4.7.3.

But in that homebrew issue I linked they point to https://bugs.mageia.org/show_bug.cgi?id=10899, which says that the problem occurs if you don't configure gcc with --enable-libstdcxx-time=yes when compiling gcc. I checked gcc -v, and indeed the version on drake006 is compiled with that option, whereas the version on my machine is not.

tkoolen, 2016-01-25:
Given this bug, https://github.com/RobotLocomotion/drake/issues/1649, and the ICE on drake006 with my latest pull request (https://github.com/RobotLocomotion/drake/pull/1648, log: http://drake006.csail.mit.edu/drake/pull-tkoolen-tk-quaternion-autodiff-into-RobotLocomotion-master), I'm a proponent of moving to 4.9. We can probably work around all of those issues, but it costs time now and in the future.

psiorx, 2016-01-25:
> I'm a proponent of moving to 4.9. 

:+1:

RussTedrake, 2016-01-25:
The reason that we have not done this yet is because it’s harder to install on Ubuntu 12.04 (not in the default apt package managers).  Should we rule that this is ok?

tkoolen, 2016-01-25:
In my opinion, it's fine. That reminds me though, we should remember to
update the travis_pre_build.sh script if we do switch.

rdeits, 2016-01-25:
Note that we're already adding the `ubuntu-toolchain-r/test` ppa on Ubuntu 12.04 on Travis in order to get gcc 4.8, so requiring 4.9 instead isn't a big change for Travis at least.

rdeits, 2016-01-26:
Fun fact (that just cost me a few hours): the `<regex>` library can be included and linked with gcc 4.8, but most of the functions aren't implemented or just silently do the wrong thing. For example, `regex_replace` just returns an empty string regardless of input. Turns out that library was never finished in 4.8 and the developers just put stubs in without any implementation. Sigh. 

See: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53631 http://stackoverflow.com/a/12665408

So...I guess I'd vote for 4.9

bradking, 2016-01-27:
For reference, the GCC project status for C++ language implementation is documented for each standard level:

* [c++11](https://gcc.gnu.org/projects/cxx0x.html)
* [c++14](https://gcc.gnu.org/projects/cxx1y.html)

The C++11 language table leaves off at GCC 4.8 and the C++14 language table essentially starts at GCC 4.9.

The libstdc++ c++11 status is documented for each version of GCC:

* [4.9.3](https://gcc.gnu.org/onlinedocs/gcc-4.9.3/libstdc++/manual/manual/status.html#status.iso.2011)
* [4.8.5](https://gcc.gnu.org/onlinedocs/gcc-4.8.5/libstdc++/manual/manual/status.html#status.iso.2011)
* [4.7.4](https://gcc.gnu.org/onlinedocs/gcc-4.7.4/libstdc++/manual/manual/status.html#status.iso.2011)

For example, we can see that regex support is documented for 4.9.3 but not 4.8.5.

The tables indicate that C++11 support is largely complete as of 4.9 and may not be in earlier versions.

RussTedrake, 2016-01-28:
ok.  let's make a decision on this.  @bradking, @billhoffman -- Do we think it's a reasonable burden to ask ubuntu folks (that main platform on which people use gcc for us) to install 4.9?

bradking, 2016-01-28:
According to [here](https://wiki.ubuntu.com/ToolChain#PPA_packages) and [here](https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test) one can install GCC 4.9 on Ubuntu via:

    sudo add-apt-repository ppa:ubuntu-toolchain-r/test
    sudo apt-get update
    sudo apt-get install g++-4.9

That is not very hard to do.  I just tried it in fresh VM installs of Ubuntu 14 and 12 and it worked for both.  Additional steps would be needed to convert the default system compiler (e.g. `/usr/bin/gcc`) to be the newer version, but that would be an invasive requirement.  Fortunately with CMake one can do just

    export CC=gcc-4.9 CXX=g++-4.9

before configuring/building to specify use of these compilers.

I think this is a reasonable requirement given that we want to use C++11, and over time it will go away.  The above steps could simply be included in drake documentation on minimum requirements somewhere.

tkoolen, 2016-01-28:
A better way in my opinion is to register both the old and new compiler with update-alternatives, making it easy to switch:
```
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-4.8 40
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-4.9 50
```
you can switch between 4.8 and 4.9 using
```
sudo update-alternatives --config g++
```

tkoolen, 2016-01-28:
(after the apt-get install of g++-4.9 of course)

rdeits, 2016-01-28:
Even better is to slave the g++ version to the gcc version, since it's very unlikely you'll want gcc and g++ to be different versions:

```
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.9 60 --slave /usr/bin/g++ g++ /usr/bin/g++-4.9 
```

tkoolen, 2016-01-28:
cool, didn't know that

bradking, 2016-01-28:
> sudo update-alternatives

This is what I meant by "additional steps", but it is an intrusive change to one's system and should be used only as a personal choice for convenience.  If I were evaluating software and one of its requirements was "change your system compiler" I might be scared off.  We should document how to build Drake without having to do that.  Using `CC=... CXX=...` is a regular part of my workflow for choosing compilers.

tkoolen, 2016-01-28:
Good point.

rdeits, 2016-01-29:
Just for future reference, if you *do* want to permanently switch your gcc and g++ versions (which, as @bradking points out, should not be mandatory), the full set of steps is:

```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-4.9 g++-4.9
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.9 60 --slave /usr/bin/g++ g++ /usr/bin/g++-4.9 
sudo update-alternatives --config gcc
```

and then select version 4.9 from the menu.

RussTedrake, 2016-01-29:
A possible solution would be to recommend people add the config steps from @bradking to the CMAKE_FLAGS environment variable.

RussTedrake, 2016-02-17:
now that #1663 is resolved, let's update the minimum gcc version to 4.9.

billhoffman, 2016-03-08:
Closing this because we are not supporting 4.7 anymore.

closed 2016-03-08 by billhoffman