RobotLocomotion/drake #1652: Building master on Windows (7, 64 bit) results in link errors in drakeLCMSystem

tkoolen, 2016-01-24:
I guess drake003 is a bit behind (stuck?), so this link error snuck in.

```
"C:\Users\Twan\code\drake\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\Twan\code\drake\pod-build\drake.vcxproj" (default target) (11) ->
(CustomBuild target) ->
  LINK : fatal error LNK1181: cannot open input file '..\lib\Release\drake_lcmtypes.lib' [C:\Users\Twan\code\drake\drake\pod-build\systems\drakeLCMSystem.vcxproj] [C:\Users\Twan\c
ode\drake\pod-build\drake.vcxproj]
  LINK : fatal error LNK1181: cannot open input file '..\lib\Release\drake_lcmtypes.lib' [C:\Users\Twan\code\drake\drake\pod-build\systems\drakeLCMSystem.vcxproj] [C:\Users\Twan\c
ode\drake\pod-build\drake.vcxproj]
```

RussTedrake, 2016-01-24:
Yes.  I saw that, too.  But I don’t know how it came in…?  
It reads like the lcmtypes is missing the exports definition.  but if that was the case, it should have been happening forever… right?

henryhwu, 2016-01-26:
It doesn't seem like drake_lcmtypes.lib was being created even in older commits. I think the cause is that the code for drake_lcmtypes.dll doesn't export anything. The vc linker doesn't create .libs for libraries that don't have exports.
I wrote a quick fix https://github.com/henryhwu/drake/tree/fix-1652 that removes the linking against lcmtypes and it compiles on Windows. However, I haven't tested this on other platforms.
```
C:\Users\Henry\drake\drake\drake\pod-build\lib\Release>dumpbin /exports drake_lcmtypes.dll
Microsoft (R) COFF/PE Dumper Version 14.00.23506.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file drake_lcmtypes.dll

File Type: DLL

  Summary

        1000 .data
        1000 .gfids
        1000 .pdata
        1000 .rdata
        1000 .reloc
        1000 .rsrc
        1000 .text
```

closed 2016-01-26 by RussTedrake

tkoolen, 2016-01-26:
Awesome, thank you!

PR #1657 (2016-01-26): Fix compile due to missing drake_lcmtypes.lib
Go back to previous CMakeLists where drakeLCMSystem isn't linked to
drake_lcmtypes.
resolves #1652