RobotLocomotion/drake #1653: Move prius model to sdf.  Load prius + parking garage in sim.

RussTedrake, 2016-01-24:


RussTedrake, 2016-02-15:
make progress on this.  my open PR has pulled in @Sproulx's stata garage sdf, and updated the instructions so that people can load that and start driving in the basement.  it's harder than it looks.  (and the tire friction definitely needs work).

RussTedrake, 2016-02-15:
just increased the friction.  handles a lot better now.  btw - @sproulx, I changed the name of the garage sdf and fixed the diffuse colors (which are [0,1], not [0,255])

liangfok, 2016-04-09:
I can perform a basic simulation of the car + garage, though the collision models and dynamics need work able to perform the basic simulation. I also still need to add the LIDAR sensors. Below are some details of what I'm running and a video showing what currently happens when the car collides with a column in the parking garage.

**Branch used:** [feature/decoupled_sdf_world_and_drake_world](https://github.com/liangfok/drake/tree/feature/decoupled_sdf_world_and_drake_world) - This branch contains the debugged SDF parser and corrected Prius.sdf. It also contains the logic that allows an SDF model's world to be offset relative to the rigid body tree's world.

**Commands:**

```
$ cd drake/examples/Cars

// Start Drake's visualizer
$ ../../../build/bin/drake-visualizer

// Start the car command GUI
$ python SteeringCommandDriver.py

// Start the simulator
$ ../../pod-build/bin/carSimLCM prius/prius.sdf stata_garage_p1.sdf
``` 

**Video:**
Here is a video showing a simulation of a Prius driving around a parking garage. For reasons as yet to be determined, it flips over and falls through the ground after hitting a concrete column.

[image]

liangfok, 2016-04-10:
Now using branch [feature/prius_in_garage](https://github.com/liangfok/drake/tree/feature/prius_in_garage).

I modified carSimLCM's command line arguments to allow the specification of the car's initial pose in the world. The new way of starting the simulator is:

```
$ cd drake/examples/Cars
$ ../../pod-build/bin/carSimLCM prius/prius.sdf [x] [y] [z] [r] [p] [y] stata_garage_p1.sdf
```

To repeatedly test what happens when the car hits a column, I initialize its pose so that it's directly in front of a column. Here is the command:

```
$ ../../pod-build/bin/carSimLCM prius/prius.sdf 3.2 3.75 0 0 0 0 stata_garage_p1.sdf
```

And here's what it initially looks like:

[image]

And here's what happens when I drive forward:

[image]

As expected, the car drives straight through the concrete column. This is because collision models currently only exist for the wheels.

liangfok, 2016-04-10:
Continuing to use branch [feature/prius_in_garage](https://github.com/liangfok/drake/tree/feature/prius_in_garage).

I configure the Prius' initial pose so that when it drives forward, its front right wheel will impact a concrete column. Here is the command:

```
$ ../../pod-build/bin/carSimLCM prius/prius.sdf 3.2 4.2 0 0 0 0 stata_garage_p1.sdf
```

Here's a video showing a brief impact, but then the wheel appears to slip to the right of the concrete column allowing the car to continue to drive through the column.

[image]

Here's a video showing a slower speed impact:

[image]

RussTedrake, 2016-05-25:
@liangfok -- I believe that you've completed this?

liangfok, 2016-05-25:
Yes, we can load the Prius SDF and URDF along with the Stata garage in Drake. I did not close the issue yet due to the poor collision modeling within the Prius. I created https://github.com/RobotLocomotion/drake/issues/2450 and will close this issue.

closed 2016-05-25 by liangfok