RobotLocomotion/drake #1654: add sensors to prius.sdf

RussTedrake, 2016-01-24:
(anticipated) lidar first please

liangfok, 2016-04-20:
I added a version of the Prius with four LIDAR sensors in this commit: https://github.com/liangfok/drake/commit/ee35f39b573bd1893371b598731b130aaf69a8e1.

Here's what it looks like in Gazebo:

[image]

[image]

I haven't tested it in Drake yet.

liangfok, 2016-04-20:
The Prius with LIDAR sensors can load inside Drake's visualizer.

[image]

However, Drake's currently doesn't parse the `sensor` element in SDF models so the sensor isn't actually being simulated.

liangfok, 2016-04-20:
I stand corrected. Drake does in fact parse LIDAR sensors in SDF files, see [RigidBodySystem.cpp:644](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.cpp#L644) . I verified that the senors are being simulated by printing the number of outputs of the rigid body system and seeing that it is 2587.

I was confused by the lack of visualization of the LIDAR sensor's rays in the Drake visualizer.

liangfok, 2016-04-21:
@RussTedrake and @david-german-tri: I've hit a roadblock. Hopefully it's just me not fully understanding System 1.0's API. I've created a new class called [SensorVisualizerLidar](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/sensor_visualizer_lidar.h) that I've cascaded at the end of the original car demo (see [this](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/examples/Cars/simulateLCMandROS.cpp#L194) code). I'm simulating a car with 4 LIDAR sensors and a total of 2587 output variables (for details, see my previous post above). Using print statements, I've verified that the input to `SensorVisualizerLidar` is indeed a vector of length 2587. My question is, how can I obtain information about the semantics of the sensor data contained within this input vector? I need to know things like the ranges of indices within this input vector for each sensor, the sensor type, and [sensor-specific parameters](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/examples/Cars/prius/prius_with_lidar.sdf#L572) like horizontal scan range, etc. 

I know joint state semantics can be determined using information contained within the `RigidBodyTree`, but I'm not sure how to do this for the sensor data. One option would be to gain access to [RigidBodySystems::sensors](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/RigidBodySystem.h#L244), which is currently private. Should I provide an accessor to `RigidBodySystem::sensors` to get the semantics of the sensor data within `SensorVisualizerLidar`'s input vector?

RussTedrake, 2016-04-21:
I think that the output frame of the sensor system should be a signal that knows how to encode itself as a ros/lcm/etc message.  The semantics of the signals will be exposed in David’s class design, but i don’t think you should be blocked on that for this application.

Also, make sure you’ve looked at lidarTest.cpp and the quadrotor example to see how they’ve been visualized so far.

liangfok, 2016-04-21:
Looking at [lidarTest.cpp](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/test/lidarTest.cpp),  the semantics of the sensor data appear to be hard-coded in a manner that's external to the output signal of the `RigidBodySystem`. The following code begins on [line 28](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/test/lidarTest.cpp#L28):

```
  // these parameters must match the sdf (i've also implicitly hard-coded the
  // box geometry from the sdf)
  const double min_yaw = -1.7;
  const double max_yaw = 1.7;
  const double max_range = 25.0;
  const double tol = 0.01;
```
The above hard-coded values are then used in the creation of a `bot_core::planar_lidar_t` on [line 58](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/test/lidarTest.cpp#L58):

```
  // publish lidar message
  bot_core::planar_lidar_t msg;
  msg.utime = -1;
  msg.nintensities = 0;
  msg.rad0 = min_yaw;
  msg.radstep = (max_yaw - min_yaw) / (distances.size() - 1);

  msg.nranges = distances.size();
  for (int i = 0; i < distances.size(); i++)
    msg.ranges.push_back(distances(i));
```

I believe the output signal of the system is just an Eigen vector and thus contains no semantic information about the sensor data. This is based on the definition of [OutputVector](https://github.com/liangfok/drake/blob/feature/add_sdf_sensors/drake/systems/plants/RigidBodySystem.h#L141) in `RigidBodySystem`, which I've replicated below:

```
  template <typename ScalarType>
  using OutputVector = Eigen::Matrix<ScalarType, Eigen::Dynamic, 1>;
```

liangfok, 2016-04-21:
Looking at the quadcopter demo's [runDynamics.cpp](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Quadrotor/runDynamics.cpp), I see that a LIDAR sensor is being created on [line 67](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Quadrotor/runDynamics.cpp#L67) in a manner that's external to the SDF:

```
  auto lidar2D = make_shared<RigidBodyDepthSensor>(
      *rigid_body_sys, "lidar2D", tree->findFrame("laser"), num_lidar_points,
      -3 * M_PI / 4, 3 * M_PI / 4, 30.0);
```

Unfortunately, I'm not seeing any code that publishes the LIDAR visualization. I'm also not actually seeing the LIDAR visualization in `drake-visualizer`:

[image]

I see a [QuadrotorOutput](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Quadrotor/QuadrotorOutput.h) that contains some LIDAR state and is being used as the output of a [Gain](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/LinearSystem.h#L105) system but I'm not seeing how the LIDAR state is being published over LCM.

Could it be that I'm not using the right version of Drake? I'm currently synced with the latest on the master.

psiorx, 2016-04-21:
The visualization was a custom Drake Director Python script that directly subscribed to and interpreted the quadrotor output  LCM type.  This was only meant to be a temporary solution for debugging.

I think the better way to do this is to have the lidar output be published in its own lidar specific LCM type that can be visualised in a more general way.

liangfok, 2016-04-21:
I curious to see the custom Director python script. Can you send it to me?

psiorx, 2016-04-21:
I don't have access to it at the moment but it was fairly simple and short.   I can get it for you when I'm back on campus.

@patmarion might be able to help if you need something sooner.

patmarion, 2016-04-21:
The drake-visualizer program currently has support to draw the hard-coded lidar published from lidarTest.cpp, you don't need a custom python script for that.  I think a publisher on the drake simulation side is yet to be designed for publishing general sensor pose and scan data.  I'm happy to help add the matching message subscriber to drake-visualizer once the publishing mechanism has been designed.  For some historical discussion, please see https://github.com/RobotLocomotion/drake/pull/1699

jwnimmer-tri, 2016-12-22:
@liangfok Done?

jwnimmer-tri, 2018-06-09:
https://github.com/RobotLocomotion/drake/blob/master/automotive/models/prius/prius_with_lidar.sdf

closed 2018-06-09 by jwnimmer-tri