RobotLocomotion/drake #1661: lcm does not compile the python bindings on windows

RussTedrake, 2016-01-26:
right in the CMakeLists.txt we have
```
find_package(PythonInterp)
if (PYTHONINTERP_FOUND AND NOT WIN32)
  add_subdirectory(lcm-python)
endif()
```
It means that we can't do the steering interface on windows.

patmarion, 2016-01-26:
I encountered this when building Director on Windows.  To make the package, I manually copied the generated .py lcm files from a different platform's build directory.

RussTedrake, 2016-01-27:
did a little bit of work on this.  Here's where I got:
- (obviously) had to remove the if (NOT WIN32) in lcm-1.0.0/CMakeLists.txt surrounding the lcm-python include.
- had to set some environment variables:
  - GLIB_PATH=drake-distro/externals/gtk/gtk3
  - VS90COMNTOOLS=$VS140COMNTOOLS
- had to update a few paths in the setup.py file, and a few quotes in the lcm-pythong/cmakelist.txt  (I will push these to the lcm cmake branch)

that got it almost working, but I ended up with a path issue
```
CustomBuild:
  running build
  running build_py
  running build_ext
  running install
  running install_lib
  creating Z:russt
  creating Z:russt\drake-distro-win
  creating Z:russt\drake-distro-win\build
  creating Z:russt\drake-distro-win\build\Lib
  creating Z:russt\drake-distro-win\build\Lib\site-packages
  creating Z:russt\drake-distro-win\build\Lib\site-packages\lcm
  copying Z:/russt/drake-distro-win/externals/lcm/lcm-1.0.0/pod-build/lcm-python\lib.win-amd64-2.7\lcm\__init__.py -> Z:russt\drake-distro-win\build\Lib\site-packages\lcm
  copying Z:/russt/drake-distro-win/externals/lcm/lcm-1.0.0/pod-build/lcm-python\lib.win-amd64-2.7\lcm\_lcm.pyd -> Z:russt\drake-distro-win\build\Lib\site-packages\lcm
  byte-compiling Z:russt\drake-distro-win\build\Lib\site-packages\lcm\__init__.py to __init__.pyc
  running install_egg_info
  Writing Z:russt\drake-distro-win\build\Lib\site-packages\lcm-1.0.0-py2.7.egg-info
```
(note the missing \ between z: and russt)

billhoffman, 2016-02-04:
I have a PR that fixes this:
https://github.com/RobotLocomotion/lcm-pod/pull/10
To fix the z:/  you have to give the path in windows \ format.  I also added some stuff that will set these automatically for you:
GLIB_PATH=drake-distro/externals/gtk/gtk3
VS90COMNTOOLS=$VS140COMNTOOLS

closed 2016-02-10 by billhoffman