RobotLocomotion/drake #1662: make -j in drake-distro produces infinitely many prompts for gurobi login credentials

rdeits, 2016-01-26:
Output of `make -j`:

```
Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com: Login failed.  Please try again.
Enter your user email for gurobi.com: Enter your password for gurobi.com:
```

tkoolen, 2016-01-27:
Related: https://github.com/RobotLocomotion/drake/issues/1392

jhoare, 2016-03-04:
The issue here appears to be when you invoke ``make -j`` it will spawn a number of processes that do not have stdin hooked up. This output is what will happen when you run ``./download_gurobi.pl < /dev/null`` for example. When ``make`` (without the ``-j`` is invoked, the only process that runs has stdin hooked up correctly. 

The most straightforward workaround that I can see is to run ``make gurobi`` first and then do a ``make -j`` afterwards.

tkoolen, 2016-03-04:
> The most straightforward workaround that I can see is to run make gurobi first and then do a make -j afterwards.

Yeah, that's what I've resorted to. I really wish we had a better solution to the question of how to handle the gurobi download though.

Related: https://github.com/RobotLocomotion/drake/issues/1523

jhoare, 2016-03-04:
Another possibility could be to have the download_gurobi.pl pop open a graphical window (assuming ti can) to ask for the username/password, and have it fall back to terminal i/o if the graphical window is not possible.

jhoare, 2016-03-04:
I decided to go ahead and take an attempt at the graphical window approach, see RobotLocomotion/gurobi#10

closed 2016-03-05 by RussTedrake

rdeits, 2016-04-25:
@jhoare I think perl-tk will need to be added to the suggested apt-get packages: http://drake002.csail.mit.edu/drake/sphinx/ubuntu.html since your fix doesn't work without it