RobotLocomotion/drake #1663: gcc 4.9 appears to break C++ LCM types in Drake

rdeits, 2016-01-26:
I just built drake master with gcc 4.9, and it appears to have broken LCM compatibility between C++ and Java (and thus Matlab) LCM types. Specifically, messages encoded from C++ have fingerprints that do not match those of their Java counterparts. I've now encountered this error on two different Ubuntu 14.04 machines. I had initially thought it was a Java version issue, but I'm now sure that gcc 4.9 is the culprit. 

To reproduce, I did the following:
* use `update-alternatives` to switch to gcc-4.8
* nuke `build`, `pod-build`, `drake/pod-build` and build everything
* run `examples/runAtlasWalking` -> everything works
* use `update-alterhatives` to switch to gcc-4.9
* nuke `build`, `pod-build`, `drake/pod-build` and build everything
* run `examples/runAtlasWalking` -> fails with:

```
Error using bipedControllers.BipedPlanEvalAndControlSystem/output (line 78)
Java exception occurred: 
java.io.IOException: LCM Decode error: bad fingerprint
        at drake.lcmt_qp_controller_input.<init>(lcmt_qp_controller_input.java:110)
        at drake.lcmt_qp_controller_input.<init>(lcmt_qp_controller_input.java:104)
```

That error occurs because the Atlas and Val demos pass the binary encoded LCM message from C++ to Java. That should always work, because it's exactly what happens when an LCM message is sent over the wire. But it's failing under gcc-4.9 because the C++ LCM types are no longer compatible.

I can also see the issue by opening a bot-spy (which uses the Java bindings) and trying to look at an LCM message published from C++. Under gcc-4.8, bot-spy works fine, but with gcc-4.9 it is unable to decode messages coming from C++. 

Has anyone successfully run the Atlas walking examples using gcc-4.9?

rdeits, 2016-01-27:
This is definitely an LCM issue, and I've traced back to g++ performing an optimization which results in a 'wrong' hash value when the -O3 flag is present (compared to the value I get with -O0, which matches Python and Java).

rdeits, 2016-01-28:
This has been fixed in upstream LCM: https://github.com/lcm-proj/lcm/pull/31

That's probably a pretty compelling reason to upgrade our LCM version in drake.

RussTedrake, 2016-02-04:
it seems that this has just become urgent.  
I think we either need to: 
1) merge our cmake lcm (from 1.0.0) into lcm 1.3 -- this would be the best solution.
2) have our superbuild use lcm 1.3 on linux/mac and lcm 1.0.0 on windows (the reason I switched to cmake).  this can be done with just pointing to different branches on our lcm-pod.
The question is... how much work would it be to try 1?

patmarion, 2016-02-04:
@jhoare the upstream lcm project code officially supports autoconf builds.  Drake actually clones lcm from here:

https://github.com/RobotLocomotion/lcm-pod

and checks out the branch named "cmake" which has a cmakeified build of lcm, primarily to support Windows.  I think Russ is suggesting you could try upgraded that branch to LCM source version 1.3 and updating the CMakeLists.txt as necessary.  I don't think anyone has looked into how much work that will require, hopefully only minimal changes required.

jhoare, 2016-02-04:
Alright I'll look into this.

billhoffman, 2016-02-05:
@RussTedrake why do you say we need: " lcm 1.0.0 on windows "?

RussTedrake, 2016-02-05:
In order to get lcm compiling on windows, I forked the lcm codebase (back at version 1.0.0) and rewrote the build system with cmake.  

lcm has continued to evolve, so our branches have diverged.  the best solution would be to merge lcm 1.3.0 with the cmake branch, and even push it upstream to the lcm developers — who are our friends.

in the lcm-pod repository you will see that the master branch is a thin wrapper that downloads lcm, sets the install prefix, and runs autoconf.  I suspect that getting that to work with lcm-1.3 should be as easy as updating the download url and unpack directory name.  the cmake branch is my fork.  

so, I think that one thing which could work quickly would be to update master to use lcm-1.3, and in the drake superbuild set things to use the master for linux/mac and the cmake branch for windows.  that would get us unstuck.

one last note: there ARE instructions in lcm for building on windows in their public distro, but they weren’t something that we could build automatically from the command line.  And when I casually asked the lcm maintainers if they’d consider accepting a cmake version, they were positive about it.

jhoare, 2016-02-05:
My plan right now is to attempt to get the upstream version of LCM to compile using CMake instead of autotools, and provide a pull request to them for that. Afterwards, I can make a modification of lcm-pod to use this cmake version of pods. I'm doing this by building off the work you've already done to get lcm version 1.0.0 to compile with cmake. 

Hows that sound?

RussTedrake, 2016-02-05:
Yes, that's ideal.  I want to have a stable soln by wednesday (running a demo where people will install), so let's touch base monday in case it turns out to be hard?

billhoffman, 2016-02-05:
@jhoare as we discussed, first get the current CMakeLists.txt file working with 1.3.0 and get that working and merged.  Then work on creating a standalone CMakeLists.txt that can be shared upstream.

jhoare, 2016-02-08:
Pull Request: #1698 For drake and RobotLocomotion/lcm-pod#13

RussTedrake, 2016-02-12:
@rdeits — can you confirm that this is fixed?

rdeits, 2016-02-16:
Confirmed! I can finally run Atlas on Ubuntu again.

closed 2016-02-16 by rdeits