RobotLocomotion/drake #1665: optimization problem without dynamic allocation on problem_type

RussTedrake, 2016-01-28:
it occurred to me that we could avoid all of the (albeit very small) dynamic allocation in the problem_type logic by simply creating const static variables for each of the problem_types and passing references to them around.  We are only using it for dynamic dispatch in the problem type hierarchy; those classes do not contain any data.

RussTedrake, 2016-01-28:
see also #1624

ggould-tri, 2016-05-25:
It is likely that the dynamic allocation is minimal (or indeed optimized out) at compile time anyway, as the classes have no constructor/destructor logic or interesting members.  Absent evidence of a cost, I am moving this to backlog.

ggould-tri, 2016-09-08:
Closing this issue as the rewritten MathematicalProgram implementation (dc05f230) has the object instantiated only once.

closed 2016-09-08 by ggould-tri