RobotLocomotion/drake #1667: cleanup / finish snopt c++ wrapper

RussTedrake, 2016-01-28:
the SNOPTWrapper.cpp is almost fully functional, but 
- does some unnecessary dynamic allocation on every call (there should not be a max number of variables, but we can simply grow the dynamic allocation on any call that exceeds the currently allocated memory).  
- it is currently not thread-safe (OK for now, but something to think about)
- it does not yet have a way to set the snopt parameters (see e.g. https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/NonlinearProgram.m#L777 for the matlab implementation)
- it does not yet return the full output from the solver.  see the matlab implementation for how we have done it in the past:   e.g. objective value, snopt info, and even infeasible constraint names.  https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/NonlinearProgram.m#L1269

RussTedrake, 2016-01-28:
#1647 is also relevant.

RussTedrake, 2018-03-13:
@jwnimmer-tri -- is this still relevant given your recent snopt backend work?

jwnimmer-tri, 2018-03-13:
The original posting sounds like quality-of-implementation concerns.  I haven't fixed anything in those areas.  My only changes along those lines have been fixing a type-checking oversight or two.

jwnimmer-tri, 2018-07-06:
I suspect that to the extent this issue is still relevant, it should be assigned to @hongkai-dai.

hongkai-dai, 2018-11-01:
Part of the "multiple threads" issue will be solved in #9896, by using the Fortran interface in snopt 7.4

hongkai-dai, 2018-11-01:
The other issues "returning full output" is captured in #9633 
We can set the snopt solver options in the current master https://github.com/RobotLocomotion/drake/blob/c922a90c101f90d807adea38120004559192c7fe/solvers/mathematical_program.h#L2273-L2282

jwnimmer-tri, 2018-11-06:
If every point already has a different individual tracking issue, perhaps this ticket (which seems like just an older TODO list) can be closed?

closed 2018-11-06 by hongkai-dai