RobotLocomotion/drake #1668: implement bindings to the Flexible Collision Library

RussTedrake, 2016-01-28:
or decide it's a bad idea
http://gamma.cs.unc.edu/FCL/fcl_docs/webpage/generated/index.html

one possible concern is a lack of support for multi-contact (E.g. if two bodies are in contact, is there a way to request the set of points at which force should be applied?)

RussTedrake, 2016-02-08:
Have been having a great exchange with Dinesh Manocha, @panjia1983, and @sherm1 about this.  In particular, @sherm1 points out that “One of Jeongseok Lee’s recent changes to FCL was to fix the API to allow it to return multiple contact points (specifically for use in physics-based simulation), although it appears to be implemented only for box-box collisions so far.”  https://github.com/flexible-collision-library/fcl/pull/52

Also noticed that DART wraps both bullet and FCL in their interface:
http://dartsim.github.io/dart/da/dcd/namespacedart_1_1collision.html

law12019, 2016-02-09:
I have been looking at the FCL API.  It appears that the collision detection will return multiple points of collision with penetration depth.  I cannot find any support for margins or intersection of geometry with rays.

billhoffman, 2016-02-10:
Add FCL as an external. @law12019

RussTedrake, 2016-02-10:
the matlab-based test with height-map collision dynamics is here:
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/test/terrainTest.m

law12019, 2016-02-24:
I added FCL and dependencies as externals included in the drake build.
I stubbed out the class FCCModel.  
I need to verify the build on windows.
I am writing tests for methods getTerrainContactPoints and collidingPoints.

law12019, 2016-03-02:
I am working on duplicating systems/plant/collision/test/ModelTest.cpp for FCL.  I have found a way to implement most of the Model methods using FCL, including closest distance and margin.  I now have enough dedicated time to make real progress on this issue.  My changes are isolated and stable so I can check them in (create a pull request) at any time if anyone wants to start working with FCL.

sherm1, 2016-03-02:
Would be great to have FCL integrated with Drake before we start making big changes to FCL this summer.

billhoffman, 2016-03-02:
One issue @RussTedrake FCL depends on boost, so it would bring in a boost depend. :(

sherm1, 2016-03-03:
> FCL depends on boost, so it would bring in a boost depend. :(

This seems like a good early project for our FCL work! ~~boost~~

RussTedrake, 2016-03-03:
That’s definitely unfortunate. Could make it hard to interface with matlab.

rpoyner-tri, 2016-03-03:
Is the matlab/boost issue unsolvable, or just unsolved? Seems to me that many apparent problems with c++ migration would be mitigated if drake could use boost without restriction. #1718, for example...

RussTedrake, 2016-03-03:
The problem is that matlab ships it's own boost libraries (but not headers).  The version changes between matlab versions.  And that library seems to get forcefully inserted before everything else in the library path.

We lost many hours to this in the past, and decided it was basically unsolvable if we aim to support a broad matlab user base.  At least for parts of the library that get loaded as a mex file in matlab.

And in almost every case, the things we want from boost are on their way into the c++ std.

rpoyner-tri, 2016-03-03:
Are we bound by performance issues run run inside matlab-controlled processes, or is that just the way things have evolved?

billhoffman, 2016-03-03:
I think the viz group at Kitware solved the boost matlab problem.  They #define namespaced boost so that there were no symbol conflicts.  I will ask around and see how hard that was to setup.

jpieper-tri, 2016-03-03:
While most of boost you want is on the way into the c++ standard, things like the filesystem TS which at best will make C++17, may still be several years before they see wide deployment across all platforms and compilers.  That's a long time to pay the penalty for re-implementation and workarounds.

billhoffman, 2016-03-03:
So, we could build our own boost using BCP:
http://www.boost.org/doc/libs/1_60_0/tools/bcp/doc/html/index.html

We could build FCL with this extracting out the few boost libraries it uses and giving it a namespace.

law12019, 2016-03-09:
The DrakeCollision FCLModel class is mostly working in my tests.  However, FCL is not computing the second collision point of the pair or the collision depth correctly.  I am working to find FCL options that will correct this.

billhoffman, 2016-03-09:
@law12019 can you post your recent results here with the duplicated test results.  Thanks.

law12019, 2016-03-09:
When two elements are not in collision, fcl::distance produces the same result as Bullet

My collision test example is two axis aligned cubes, sized 20 and 10 intersecting at the corner. The smaller cube is translated (14,14,14). The two deepest penetrating points on the two elements in world coordinate system are (20,20,20) and (19,19,19), in model coordinate system: (10,10,10) and (-5,-5,-5).

DrakeCollision::BulletModel results:
1 collision point pair:
  { distance: -1
    pointA : (10, 9.333, 9.555)
    pointB : (-5, -4.6666, -4.4444)
    normal : (-1, 0, 0) }

FCL results (points in world coordinate system):
20 collision points:
in collision 20:
    0:p=(10,-10,10)n=(-0,-0,1)d=1,
    1:p=(10,10,10)n=(-0,-0,1)d=1,
    2:p=(10,10,10)n=(-0,1,-0)d=1,
    3:p=(10,10,10)n=(-0,1,0)d=1,
    4:p=(10,-10,10)n=(1,-0,-0)d=1,
    5:p=(10,10,10)n=(1,-0,-0)d=1,
    6:p=(-10,10,10)n=(-0,1,-0)d=1,
    7:p=(10,10,10)n=(-0,1,-0)d=1,
    8:p=(-10,10,10)n=(-0,1,0)d=1,
    9:p=(10,10,10)n=(-0,1,0)d=1,
    10:p=(10,10,10)n=(1,-0,-0)d=1,
    11:p=(10,10,-10)n=(1,-0,-0)d=1,
    12:p=(10,10,10)n=(-0,-0,1)d=1,
    13:p=(-10,10,10)n=(-0,-0,1)d=1,
    14:p=(10,10,10)n=(-0,-0,1)d=1,
    15:p=(-10,10,10)n=(-0,-0,1)d=1,
    16:p=(10,10,10)n=(-0,-0,1)d=1,
    17:p=(10,10,10)n=(-0,1,0)d=1,
    18:p=(10,10,-10)n=(-0,1,0)d=1,
    19:p=(10,10,10)n=(-0,-0,1)d=1,

I see two problems.  First, I do not have enough information to compute the second point of the DrakeCollision::PointPair. Although it will not produce identical results, I am use p- n*d.

Second issue is that some of the collision points look wrong: (-10,10,10), (10,-10,10) (10,10,-10)

I am going to test different shapes and mesh and then submit the pull request.

law12019, 2016-03-18:
I created a pull request with the initial implementation.

sherm1, 2016-03-28:
@law12019, as of this weekend the FCL upstream master branch is boost-free for the library, although it is still in use for the test cases.

RussTedrake, 2018-10-18:
i think it's safe to close this one.

closed 2018-10-18 by RussTedrake