RobotLocomotion/drake #1672: Implement Matlab-free InstantaneousQPController

rdeits, 2016-01-29:
I've been working on this for a while, but I wanted to create an issue to track progress.

The goal is to run a standalone version of the InstantaneousQPController entirely in C++, using only the robot URDF and configuration YAML files. This is necessary for control on Valkyrie, since we won't have Matlab onboard the robot, and it's generally necessary for the cleanup of Drake. 

When finished, we will have a new InstantaneousQPController class in C++, which can be constructed from a URDF filename, YAML config filename, and a few other options (like whether to add hands). This controller will be able to run as a standalone LCM node or run synchronously as part of a Drake system.

rdeits, 2016-01-29:
I'm actually pretty close to being done with this. The new controller class has been created, and it handles both Atlas and Valkyrie correctly. It currently fails on a few of the Atlas unit tests because there's not yet a way to add hands to a C++ RBT model, because `RigidBodyTree::addRobotFromURDF` does not support welding the new robot to a specific link of the existing robot. I'll try to add support for that now.

rdeits, 2016-02-02:
@RussTedrake I'm currently feeling stuck on the last stage of this because I no longer feel like I understand what the interface should be, if we're going to be supporting multiple message protocols/transports. Once I know what the interface should be, I can move InstQP entirely to C++, provide Swig bindings for Matlab if needed, and be done.

RussTedrake, 2016-02-27:
this is in.

closed 2016-02-27 by RussTedrake