RobotLocomotion/drake #1673: Drake, compile error, it can not find VTK

javieralonso, 2016-01-29:
Hi, in my Mac, I followed the instructions to install dependencies, when invoking make I get the following error:

CMake Error at src/vtk/CMakeLists.txt:1 (find_package):
  By not providing "FindVTK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VTK", but
  CMake did not find one.

  Could not find a package configuration file provided by "VTK" with any of
  the following names:

    VTKConfig.cmake
    vtk-config.cmake

  Add the installation prefix of "VTK" to CMAKE_PREFIX_PATH or set "VTK_DIR"
  to a directory containing one of the above files.  If "VTK" provides a
  separate development package or SDK, be sure it has been installed.

Drake contains some VTK folders but none of those files.

Thanks!

RussTedrake, 2016-01-29:
I assume this happened while it was building the director external project?

javieralonso, 2016-01-29:
[ 65%] Completed 'ctkPythonConsole'
[ 75%] Built target ctkPythonConsole
[ 78%] Performing configure step for 'director'
loading initial cache file /Users/jalonso/MIT/toyotaAV/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/director-cache-Release.cmake

Shall I install VTK independently?

RussTedrake, 2016-01-30:
I suspect @patmarion will have some fast advice for you

patmarion, 2016-01-30:
Have you installed vtk5 with homebrew?  You can install that dependency with these commands:

    brew tap homebrew/science
    brew tap homebrew/python
    brew install python numpy
    brew install vtk5 --with-qt

If you have already done that and it is still not working, please verify this file exist on your filesystem:

/usr/local/opt/vtk5/lib/vtk-5.10/VTKConfig.cmake

That's the file that should be located automatically if it has been installed.

javieralonso, 2016-01-30:
Thanks, I am debugging that now. It is not happy:
[ 63%] Built target vtkGraphicsPythonD
make: *** [all] Error 2
Warning: It appears you have MacPorts or Fink installed.
Software installed with other package managers causes known problems for
Homebrew. If a formula fails to build, uninstall MacPorts/Fink and try again.
READ THIS: https://git.io/brew-troubleshooting

javieralonso, 2016-01-30:
Pat, did you have trouble like this one with brew install vtk5 --with-qt before?

patmarion, 2016-02-01:
No.  If homebrew has an error while installing software it can be difficult to troubleshoot, it usually means you have some other non-homebrew software installed that is conflicting.

chuckyouliu, 2016-02-04:
@javieralonso did you ever resolve this? Had the same error but don't have MacPorts or Finks installed

patmarion, 2016-02-04:
I recently made a homebrew bottle for vtk with qt.  You can attempt to install this if you have mac osx mavericks or newer:

wget https://dl.dropboxusercontent.com/s/mtacwrjgmtanfaz/vtk5-5.10.1_2.mavericks.bottle.1.tar.gz
brew tap homebrew/science
brew install qt
brew install vtk5-5.10.1_2.mavericks.bottle.1.tar.gz

That should install vtk with qt, and does not require a source compile.  Afterward, sure you run drake's install_prereqs.sh script (after make download-all) and that it runs to completion successfully.  If the above does not work, you can undo it with: brew uninstall vtk5

chuckyouliu, 2016-02-05:
this worked, thanks!

javieralonso, 2016-02-05:
> Afterward, make sure you run drake's install_prereqs.sh script (after make

Thank you. I could install VTK with the link you provided,  the file cmake
exists, but install prerequisites still can not find it.

patmarion, 2016-02-05:
Can you post more details about which command you are running and what the output is?  Please post the full output, or link to it, so we can try to help.

javieralonso, 2016-02-05:
Thanks. This is the error:

[ 78%] Performing configure step for 'director'
loading initial cache file /Users/aaa/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/director-cache-Release.cmake
CMake Error at src/vtk/CMakeLists.txt:1 (find_package):
  By not providing "FindVTK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VTK", but
  CMake did not find one.

  Could not find a package configuration file provided by "VTK" with any of
  the following names:

    VTKConfig.cmake
    vtk-config.cmake

  Add the installation prefix of "VTK" to CMAKE_PREFIX_PATH or set "VTK_DIR"
  to a directory containing one of the above files.  If "VTK" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/Users/aaa/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/director-build/CMakeFiles/CMakeOutput.log".
make[7]: *** [src/director-stamp/director-configure] Error 1
make[6]: *** [CMakeFiles/director.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

patmarion, 2016-02-05:
Ok, so again confirm this file exists:

/usr/local/opt/vtk5/lib/vtk-5.10/VTKConfig.cmake

Then perhaps you need to trigger a reconfigure:

cd drake-distro/externals/director
make clean
make drake

javieralonso, 2016-02-05:
Thank you, that worked.
The new error I am stuck with is:

/Users/jalonso/MIT/toyotaAV/drake-distro/drake/pod-build/bindings/swig/autodiffutilsPYTHON_wrap.cxx:154:11: fatal error:
      'Python.h' file not found
# include <Python.h>
          ^
1 error generated.
make[6]: *** [bindings/swig/CMakeFiles/_autodiffPython.dir/autodiffutilsPYTHON_wrap.cxx.o] Error 1
make[5]: *** [bindings/swig/CMakeFiles/_autodiffPython.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[2]: *** [CMakeFiles/drake.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

patmarion, 2016-02-05:
Are you sure you have installed prereqs?

cd drake-distro
make download-all
./install_prereqs.sh homebrew

javieralonso, 2016-02-05:
Yes. It might be related to:

Warning: Tap homebrew/python already tapped.
Error: Formulae found in multiple taps:
 * homebrew/python/numpy
 * samueljohn/python/numpy

Please use the fully-qualified name e.g. homebrew/python/numpy to refer the formula.

patmarion, 2016-02-05:
So you are saying that "./install_prereqs.sh homebrew" fails with that error message?  You need to resolve all the issues so that command can run completely without errors.  Try running:

brew untap samueljohn/python

and then run install_prereqs again.

But that implies you may have installed some unofficial homebrew packages.  You might have conflicting packages installed (which could explain why your vtk install was failing earlier).

javieralonso, 2016-02-05:
I did so, and make clean inside the swing folder, but I am still getting the same error after make.
Here the output of install_prereqs.sh, which now only gives errors of already installed packages.
Shall I unlink all of them and try again?

31-35-78:drake-distro jalonso$ ./install_prereqs.sh homebrew
Warning: cmake-3.4.3 already installed
Error: pkg-config-0.28 already installed
To install this version, first `brew unlink pkg-config`
Error: gtk+-2.24.28_1 already installed
To install this version, first `brew unlink gtk+`
installing prereqs for drake
java version >= 1.6: YES
Warning: graphviz-2.38.0 already installed
installing prereqs for externals
Warning: cmake-3.4.3 already installed
Warning: swig-3.0.8 already installed
Error: pkg-config-0.28 already installed
To install this version, first `brew unlink pkg-config`
Error: gtk+-2.24.28_1 already installed
To install this version, first `brew unlink gtk+`
installing prereqs for /Users/jalonso/MIT/toyotaAV/drake-distro/externals/eigen
installing prereqs for /Users/jalonso/MIT/toyotaAV/drake-distro/externals/libbot
Warning: cmake-3.4.3 already installed
Warning: libpng-1.6.21 already installed
Warning: jpeg-8d already installed
Error: gtk+-2.24.28_1 already installed
To install this version, first `brew unlink gtk+`
installing prereqs for /Users/jalonso/MIT/toyotaAV/drake-distro/externals/bullet
Warning: gnu-sed-4.2.2 already installed
Error: wget-1.16.3 already installed
To install this version, first `brew unlink wget`
installing prereqs for /Users/jalonso/MIT/toyotaAV/drake-distro/externals/director/distro/pods/drake-distro
Warning: Tap homebrew/science already tapped.
Warning: Tap homebrew/python already tapped.
Warning: python-2.7.11 already installed, it's just not linked
Warning: homebrew/python/numpy-1.10.4 already installed
Warning: homebrew/science/vtk5-5.10.1_2 already installed
installing prereqs for /Users/jalonso/MIT/toyotaAV/drake-distro/externals/swig_matlab
Warning: autoconf-2.69 already installed
Warning: automake-1.15 already installed
Warning: libtool-2.4.6 already installed

patmarion, 2016-02-05:
If you carefully read that log, you'll see some some important stuff that is worth addressing.  This one in particular looks relevant to the compile error you are encountering:

 Warning: python-2.7.11 already installed, it's just not linked

You might want to resolve that (linking it)

javieralonso, 2016-02-05:
Thank you, sorry for that. The next error is mex related.
Note, I did run mex -setup in Matlab. It set it up to  'Xcode with Clang' 
I also have XCode with command line tools installed.

make gives this error:

[100%] Generating matlab/addpath_spotless.m, matlab/rmpath_spotless.m
Unrecognized MATLAB option "wait".

                      < M A T L A B (R) >
            Copyright 1984-2015 The MathWorks, Inc.
             R2015a (8.5.0.197613) 64-bit (maci64)
                       February 12, 2015

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
Toolbox "cddmex:1.0.1:maci64" added to the Matlab path.
	Academic License

 Installing SPOT in /Users/jalonso/MIT/toyotaAV/drake-distro/externals/spotless/spotless:
 updating the path...
 compiling the binaries...Building with 'Xcode with Clang'.
MEX completed successfully.
Error using mex
No supported compiler or SDK was found. For options, visit
http://www.mathworks.com/support/compilers/R2015a/maci64.html.

Error in spot_install (line 20)
mex spot_mex_msspoly_check_canonical.cpp

Error in spot_pods_install (line 5)
spot_install;

RussTedrake, 2016-02-05:
Did you also run mex -setup c++

javieralonso, 2016-02-05:
That was giving an error, now solved after following the instructions here:
http://www.mathworks.com/matlabcentral/answers/246507-why-can-t-mex-find-a-supported-compiler-in-matlab-r2015b-after-i-upgraded-to-xcode-7-0#answer_194526

Now it made successfully! Thanks for your help!

closed 2016-02-05 by javieralonso