RobotLocomotion/drake #1674: Unable to add drake path using latest build

eric-wieser, 2016-01-31:
Built from source on windows using d1d451bca787a0a2504ce47ddf396a6b52b3b1bd

```
>> run('C:\Users\wiese\Documents\Courses\6.832\drake-install\matlab\addpath_drake.m')
Out of memory. The likely cause is an infinite recursion within the program.

Error in fileparts (line 40)
if ispc
```

eric-wieser, 2016-01-31:
Ok, seems i was running the wrong file. What purpose does this file serve though? It seems like it inevitably runs itself in a recusion loop...

RussTedrake, 2016-01-31:
it is supposed to call the other addpath_drake.  somehow the install must have written the wrong path into that file?  would be good to resolve..

eric-wieser, 2016-01-31:
For me, that file contains:

```
function addpath_drake()
  mfiledir = fileparts(which(mfilename));
  wd = cd(fullfile(mfiledir,''));
  addpath_drake();
  cd(wd);
end
```

RussTedrake, 2016-01-31:
The relpath logic failed you on install.  see here: https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L229

eric-wieser, 2016-02-09:
Why do we invoke python to get relative paths instead of using `file(RELATIVE_PATH from to)`?

RussTedrake, 2016-02-09:
Some of this is old code.  I agree, it looks like using file (RELATIVE_PATH ) is better.  updating it would be great.

billhoffman, 2016-03-01:
We sort of need matlab for this one...   But we have a trial one.  Might want to look at licenses.

closed 2016-03-04 by RussTedrake