RobotLocomotion/drake #1675: Confusion over whether lcmgl can be found

eric-wieser, 2016-01-31:
Here's my session:

```
>> addpath_drake
>> testTaskSpaceRRT
 Added the lcm jar to your javaclasspath (found via cmake)
Warning: Currently all of your LCM traffic will be broadcast to the network, because:
'export' is not recognized as an internal or external command,
operable program or batch file.
 
> In checkDependency (line 89)
  In checkDependency (line 100)
  In LCMGLClient (line 3)
  In MotionPlanningTree/setLCMGL (line 305)
  In SE3MotionPlanningTree/setLCMGL (line 65)
  In SE3MotionPlanningTree (line 12)
  In TaskSpaceMotionPlanningTree (line 14)
  In testTaskSpaceRRT (line 133) 
 
 LCMGL not found.  LCMGL support will be disabled.
 To re-enable, add bot2-lcmgl.jar to your matlab classpath using javaaddpath.
 
Error using checkDependency (line 450)
Cannot find required dependency: lcmgl

Error in LCMGLClient (line 3)
  checkDependency('lcmgl');

Error in MotionPlanningTree/setLCMGL (line 305)
      obj.lcmgl = LCMGLClient(name);

Error in SE3MotionPlanningTree/setLCMGL (line 65)
      obj = setLCMGL@CompositeVertexArrayTree(obj, varargin{:});

Error in SE3MotionPlanningTree (line 12)
      obj = obj.setLCMGL('SE3MotionPlanningTree',[0, 0, 0]);

Error in TaskSpaceMotionPlanningTree (line 14)
      obj = obj@CompositeVertexArrayTree({SE3MotionPlanningTree(), ...

Error in testTaskSpaceRRT (line 133)
TA = TaskSpaceMotionPlanningTree(r, 'l_hand', point_in_link_frame);
```

Apparently, lcmgl is both `found via cmake` and also `not found`.

My guess would be the missing `export` command on windows?

RussTedrake, 2016-01-31:
There is a chance that it’s been found, and then removed from memory.  check the github issues about clear java.

eric-wieser, 2016-01-31:
Changing `systemWCmakeEnv` to:

```matlab
function [status,result] = systemWCmakeEnv(command)

% The input and output of this function are equivalent to 
% matlab's system function.  But since matlab messes with
% the shell environment (to the point where many apps fail 
% to run due to dynamic linking errors), I have the drake
% cmake script the following environment varibles:
%
%   LD_LIBRARY_PATH  (DYLD_LIBRARY_PATH and DYLD_FRAMEWORK_PATH on mac)
%
% to disk.  This function temporarily
% restores the values of those variables before calling 
% command, to help with execution.  

if (ismac)
  old_fw_path = getenv('DYLD_FRAMEWORK_PATH');
  old_lib_path = getenv('DYLD_LIBRARY_PATH');
  setenv('DYLD_FRAMEWORK_PATH',getCMakeParam('DYLD_FRAMEWORK_PATH'));
  setenv('DYLD_LIBRARY_PATH', getCMakeParam('DYLD_LIBRARY_PATH'));
  
  [status,result] = system(command);
  setenv('DYLD_FRAMEWORK_PATH', old_fw_path);
  setenv('DYLD_LIBRARY_PATH', old_lib_path);
else
  old_lib_path = getenv('LD_LIBRARY_PATH');
  setenv('LD_LIBRARY_PATH',getCMakeParam('LD_LIBRARY_PATH'));
  [status,result] = system(command);
  setenv('LD_LIBRARY_PATH', old_lib_path);
end
```

Removes the following error

> Warning: Currently all of your LCM traffic will be broadcast to the network, because:
'export' is not recognized as an internal or external command,
operable program or batch file.

but still fails the dependency check

PR #1677 (2016-02-11): Don't rely on `system('export ...; cmd')` working
Tries to fix #1675, but only does half the job
closed 2016-02-11 by RussTedrake