RobotLocomotion/drake #1676: CMake syntax errors with windows path separators

eric-wieser, 2016-01-31:
```
 CMake Warning (dev) at solvers/qpSpline/cmake_install.cmake:68 (file):
      Syntax error in cmake code at

        C:/Users/wiese/Repos/drake-distro/drake/pod-build/solvers/qpSpline/cmake_install.cmake:68

      when parsing string

        C:\Users\wiese\Repos\drake-distro\drake\pod-build\lib\pkgconfig/drakeSplineGeneration.pc

      Invalid escape sequence \U

      Policy CMP0010 is not set: Bad variable reference syntax is an error.  Run
      "cmake --help-policy CMP0010" for policy details.  Use the cmake_policy
      command to set the policy and suppress this warning.
    Call Stack (most recent call first):
      solvers/cmake_install.cmake:36 (include)
      cmake_install.cmake:1138 (include)
    This warning is for project developers.  Use -Wno-dev to suppress it.
```

eric-wieser, 2016-02-02:
Here's how this manifests itself. For me, the file `drake-distro\externals\lcm\lcm-1.0.0\pod-build\lcm\cmake_install.cmake` contains the following lines

```

if(NOT CMAKE_INSTALL_COMPONENT OR "${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified")
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include/lcm" TYPE FILE FILES
    "C:/Users/wiese/Repos/drake-distro/externals/lcm/lcm-1.0.0/lcm/eventlog.h"
    "C:/Users/wiese/Repos/drake-distro/externals/lcm/lcm-1.0.0/lcm/lcm.h"
    "C:/Users/wiese/Repos/drake-distro/externals/lcm/lcm-1.0.0/lcm/lcm_coretypes.h"
    "C:/Users/wiese/Repos/drake-distro/externals/lcm/lcm-1.0.0/lcm/lcm-cpp.hpp"
    "C:/Users/wiese/Repos/drake-distro/externals/lcm/lcm-1.0.0/lcm/lcm-cpp-impl.hpp"
    )
endif()

if(NOT CMAKE_INSTALL_COMPONENT OR "${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified")
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib/pkgconfig" TYPE FILE FILES "C:\Users\wiese\Repos\drake-distro\externals\lcm\lcm-1.0.0\pod-build\lib\pkgconfig/lcm.pc")
endif()
```

Note that somehow backslashes appeared in the paths within the second conditional block

eric-wieser, 2016-02-04:
More followup - seems that `PKG_CONFIG_OUTPUT_PATH` is tainted with the windows separators

billhoffman, 2016-02-26:
Sorry, this has taken so long.   I have not yet created a PR, but I think this will fix the issue in case anyone is stuck on it.

```
diff --git a/pods.cmake b/pods.cmake
index 1807386..a147004 100644
--- a/pods.cmake
+++ b/pods.cmake
@@ -94,9 +94,11 @@ macro(pkg_config_path var)
 endmacro()
 
 macro(shell_path var)
-  if (WIN32 AND cygpath AND ${var})
-    call_cygpath(-u ${var})
+  if (WIN32 AND ${var})
+    file(TO_CMAKE_PATH "${${var}}" ${var})
   endif()
 endmacro()
 '''

billhoffman, 2016-03-21:
https://github.com/RobotLocomotion/drake/pull/1899 should fix this.

RussTedrake, 2016-03-21:
fyi - you can use https://help.github.com/articles/closing-issues-via-commit-messages/ in your PR for things like this.

billhoffman, 2016-03-23:
Try it now with a recent master.

closed 2016-03-23 by billhoffman