RobotLocomotion/drake #1684: Coordinate frame issue in runAtlasHandControl

rdeits, 2016-02-02:
On master: 

```
Error using Trajectory/inFrame (line 69)
couldn't find a coordinate transform from the trajectory frame atlasFrames.AtlasState+right_atlasFrames.HandState+left_atlasFrames.HandState to the requested
frame atlasPosition+s-model_articulatedPosition+s-model_articulatedPosition

Error in Trajectory/subsref (line 217)
        [varargout{:}] = builtin('subsref',a,s);

Error in Visualizer/playback (line 85)
        xtraj = xtraj.inFrame(obj.getInputFrame);  % try to convert it

Error in runAtlasHandControl (line 150)
  playback(v,traj,struct('slider',true));
```

rdeits, 2016-02-02:
I don't know how this got through, but it's failing on master for me, and on the branch from #1678

RussTedrake, 2016-02-08:
I can predict your reaction, @rdeits, but the reason it got through the build servers is probably because they do not test the visualizers (they get a NullVisualizer object which has an empty playback method).  

Assigning Eric, since it was his change.

eric-wieser, 2016-02-08:
Can you point me to the blame that shows this was me?

RussTedrake, 2016-02-08:
Ah.  I might be wrong.  I thought it was related to the trajectory frame fixes you passed in earlier.

eric-wieser, 2016-02-08:
That's not unreasonable to assume - I'll take a look, if I'm able to run that on windows.

eric-wieser, 2016-02-09:
Yep, no dice - no `gurobi` for me

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri