RobotLocomotion/drake #1686: Investigate drake MATLAB depends on toolkits

billhoffman, 2016-02-03:
Look into what toolkits Kitware's MATLAB is missing.

tkoolen, 2016-02-03:
As a conservative estimate, the toolboxes we used for the EdX class were:

```
Simulink 
Communications System Toolbox
Computer Vision System Toolbox  
Control System Toolbox 
Curve Fitting Toolbox 
DSP System Toolbox 
Fixed-Point Toolbox 
HDL Coder 
Image Processing Toolbox
MATLAB Report Generator
Optimization Toolbox
Parallel Computing Toolbox 
Signal Processing Toolbox
SimEvents
Simulink 3D Animation 
Simulink Control Design
Simulink Design Optimization
Simulink Report Generator  
Simulink Verification and Validation 
Stateflow 
Symbolic Math Toolbox 
System Identification Toolbox
```

RussTedrake, 2016-02-03:
that is extremely conservative; most of those are not essential.  a better list is here: https://github.com/RobotLocomotion/drake/wiki/Installation-and-Quickstart

closed 2016-02-10 by billhoffman