RobotLocomotion/drake #1688: Jenkins https certificate invalid

tkoolen, 2016-02-03:
Clicking on the Jenkins Details link in the build results for a pull request results in:

[image]

billhoffman, 2016-02-03:
Yup, it is a self generated certificate.   @RussTedrake I think you were going to look into possibly getting a certificate from MIT, if not we would have to purchase one.  For now just click advanced and to to the site.

RussTedrake, 2016-02-03:
I believe i already sent Jamie the info for the mit bits.  - Russ

jamiesnape, 2016-02-03:
I only have information about the IP address. Nothing about an SSL certificate.

RussTedrake, 2016-02-03:
Can you follow up with @mpearrow about it?

jamiesnape, 2016-02-03:
Sure.

jamiesnape, 2016-02-04:
Certificate request is being processed now. Should be available by tomorrow.

tkoolen, 2016-02-04:
Awesome. Other than that the Jenkins build process seems to be going smoothly!

jamiesnape, 2016-02-04:
Issued and deployed.

tkoolen, 2016-02-04:
Verified fixed.

closed 2016-02-04 by tkoolen