RobotLocomotion/drake #1690: Trouble Running Visualizer and runPendulumDynamics

jhoare, 2016-02-04:
I am attempting to run the "runPendulumDynamics" example and it does not seem to be working correctly.

Here's what I am running (each in its own terminal):

```
./drake/pod-build/bin/runPendulumDynamics
./build/bin/drake-visualizer
```

When I do this the visualizer comes up, but it is just an empty ground plane. 

In the terminal that drake-visualizer was launched in, the following error scrolls out continuously:

```
Traceback (most recent call last):
  File "/home/jhoare/repos/github/drake-distro/build/lib/python2.7/dist-packages/director/lcmUtils.py", line 116, in handleMessage
    callback(msg)
  File "/home/jhoare/repos/github/drake-distro/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 309, in onViewerDraw
    link = self.getLink(robotNum, linkName)
  File "/home/jhoare/repos/github/drake-distro/build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 287, in getLink
    return self.robots[robotNum][linkName]
KeyError: 0
```

RussTedrake, 2016-02-04:
You have to start the visualizer before the simulator.  (It’s missing the load command).  
We should definitely handle that more gracefully.

jhoare, 2016-02-04:
Sorry, I was actually starting the visualizer first. I just did it again and had the same issue.

jhoare, 2016-02-04:
By the way, I'm running Ubuntu 15.10: 

```
jhoare@messier:~/repos/github/drake-distro$ gcc --version
gcc (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

RussTedrake, 2016-02-04:
@patmarion, I'm on a train.  can you take a look?

RussTedrake, 2016-02-04:
Actually, I've just merged @patmarion updated sha for the director.  Can you make sure your update pulls in the new sha and try again?

jhoare, 2016-02-04:
Director has been updated, and I now have a more meaningful error message:

```
jhoare@messier:~/repos/github/drake-distro$ ./build/bin/drake-visualizer 
error decoding message on channel: DRAKE_VIEWER_LOAD_ROBOT
Error locating link name: world
Error locating link name: base_part2
Error locating link name: arm
Error locating link name: arm_com
```

patmarion, 2016-02-04:
Haha, yes I fixed the key error spam.  Ok, so here is the problem:

error decoding message on channel: DRAKE_VIEWER_LOAD_ROBOT

I bet this is due to: https://github.com/RobotLocomotion/drake/issues/1663

To summarize, drake builds lcm 1.0 which is broken for newer versions of gcc.  If you are on ubuntu 15, you probably have gcc > 4.8.  The symptom is that certain lcm messages fail to decode.  In your case, the DRAKE_VIEWER_LOAD_ROBOT fails to  decode, but the draw messages are fine, so the viewer receives draw messages but misses the load message.

You can resolve this by upgrading to the latest version of lcm (1.3 I believe).  We hope to upgrade drake's verison of LCM very soon, but in the meantime you could just install lcm yourself and then comment out lcm from the drake build (perhaps there's an easier way).  LCM is quite easy to install to /usr/local.  Just download the latest source code release and untar, then ./configure && make install.

patmarion, 2016-02-04:
you could manually install lcm 1.3 to drake's build prefix:

wget https://github.com/lcm-proj/lcm/releases/download/v1.3.0/lcm-1.3.0.zip

...

./configure --prefix=/home/jhoare/repos/github/drake-distro/build
make install

patmarion, 2016-02-04:
after installing lcm 1.3 you'll need to make clean and re make in drake-distro/drake to regenerate the lcm types.

jhoare, 2016-02-04:
Would it be worthwhile for me to get the system to build lcm 1.3 instead? or are there other things holding that back?

RussTedrake, 2016-02-04:
That would be great.  I just put in a comment on https://github.com/RobotLocomotion/drake/issues/1663

jhoare, 2016-02-04:
I followed @patmarion's steps above to install lcm1.3 manually into drake's build prefix and it seems to work as expected now.

RussTedrake, 2016-02-08:
am closing this one.  #1663 captures what we need to do for the fix.

closed 2016-02-08 by RussTedrake