RobotLocomotion/drake #1694: Drake build issue

samkhal, 2016-02-06:
I'm building drake on Ubuntu 14.04.
```
[ 34%] Building CXX object src/app/CMakeFiles/ddApp.dir/ddLCMThread.cpp.o
make[10]: *** No rule to make target `/usr/lib/libvtkPythonCore.so.5.8.0', needed by `lib/libddApp.so'.  Stop.
make[9]: *** [src/app/CMakeFiles/ddApp.dir/all] Error 2
make[8]: *** [all] Error 2
make[7]: *** [src/director-stamp/director-build] Error 2
make[6]: *** [CMakeFiles/director.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```
Before that, there were a large number of warnings like this, referring to various files referenced by VTKTargets.cmake:

```
-- The imported target "vtkPythonCore" references the file
   "/usr/lib/libvtkPythonCore.so.5.8.0"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.
```

Thanks.

zamzami, 2016-02-07:
I had a similar problem while building on mac OSX 10.11.1 and managed to resolve it by referring to #1673. Basically you have to install independently VTK5 [http://www.vtk.org/download/](url), but there is probably a better solution.

patmarion, 2016-02-07:
You should install prereqs using drake's install_prereqs.sh script.  If you have not done that, please refer to the build instructions.  For both ubuntu 14 and mac osx, that script will take care of installing vtk.  You should not have to download and install vtk manually.

patmarion, 2016-02-07:
@samkhal it looks like you may have had vtk5 and then uninstalled it?  You should install vtk with:

sudo apt-get install libvtk5-dev libvtk5-qt4-dev python-vtk

samkhal, 2016-02-08:
Thanks @patmarion. I did have vtk installed sometime back and that may have been the culprit, since I _had_ run install_prereqs. Reinstalling vtk as you suggested seems to have fixed it.

closed 2016-02-08 by samkhal