RobotLocomotion/drake #1695: drake-visualizer quit unexpectedly

zamzami, 2016-02-07:
Drake Director crashes on a fresh build of drake on a Mac OS X 10.11.1 and Matlab 2015b. It is probably relevant to bug #1452 and issue #317 though installed JDK 8u74 via Oracle as suggested by @patmarion. 
I also set it to prefer to IPv4 as suggested in #558
LCM works correctly, tested it by running `/examples/Pendulum/testLCMPlant`.
Am I missing something? 
```
	Academic License

>> addpath_drake
>> contactImplicitBrick
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
attempting to launch the drake director
Error using BotVisualizer (line 48)
Failed to automatically start up a viewer (or to receive the ack, see https://github.com/RobotLocomotion/drake/issues/317)

Error in RigidBodyManipulator/constructVisualizer (line 1708)
            v = feval(type,arg{:});

Error in TimeSteppingRigidBodyManipulator/constructVisualizer (line 1097)
      v = constructVisualizer(obj.manip,varargin{:});

Error in contactImplicitBrick (line 24)
  v = constructVisualizer(plant_ts);
```
Same goes for `example/Quadrotor` 
``` 
>> runDircol
Elapsed time is 7.862391 seconds.
attempting to launch the drake director
Error using BotVisualizer (line 48)
Failed to automatically start up a viewer (or to receive the ack, see https://github.com/RobotLocomotion/drake/issues/317)

Error in RigidBodyManipulator/constructVisualizer (line 1708)
            v = feval(type,arg{:});

Error in runDircol (line 40)
  v = constructVisualizer(r);
```
Problem report [http://pastebin.com/KxZTGQ6R](url)

patmarion, 2016-02-07:
Thanks for posing the pastebin output, that's helpful.  It looks to me like you may have mismatched versions of Python.  This happens sometimes when you install Python after you have already compiled drake or installed the vtk dependency.  Some of your libraries are compiled against Apple's system python, and some are compiled against Homebrew's Python.  I see both in your report.  They are:

/Library/Frameworks/Python.framework
/usr/local/opt/python/Frameworks/Python.framework

You can check which version of Python your installed copy of VTK is using:

$ otool -L /usr/local/opt/vtk5/lib/vtk-5.10/libvtkPythonCore.dylib | grep "Python.framework"
	/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.0)

If it is using /Library/Frameworks/Python.framework and not Homebrew python from /usr/local then you may have to:  "brew uninstall vtk5 && brew install vtk5 --with-qt"     to re-install VTK with the correct version of Python.

If VTK is not the problem, then you might just need to do a clean build of Director:

  cd externals/director
  make clean && make drake

zamzami, 2016-02-07:
Thank you @patmarion for your prompt reply and the detailed walkthrough. Unfortunately, the problem still persists. 

I did the following 

- checked VTK python dependency and found that it is using the Hombrew version. `/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.0)`
-  Cleaned and rebuilt successfully director using `make drake-distro`  the available build options in the `Makefile` are `drc`, `drake-distro` and `superbuild`
- After restart and run  `contactImplicitBrick ` I still receive the same error and here is the new problem report [http://pastebin.com/FmYYYQye](url)

patmarion, 2016-02-07:
In another issue you said that you manually installed vtk.  Is it possible that is somehow in your path?  A few more things to check.  Please let me know the output of these commands:

    $ python
    >>> import vtk
    >>> print vtk.__file__

    $ cd drake
    $ otool -L ./build/lib/*dylib | grep Python

And does this crash:

    $ cd drake
    $ ./build/bin/directorPython
    >>> import vtk

To move forward with a working visualizer you can download the pre-compiled drake release.
You can download the drake pre-compiled release here, then run build/bin/drake-visualizer from the extracted folder:

https://github.com/RobotLocomotion/drake/releases/download/v0.9.11/drake-v0.9.11-mac.tar.gz

zamzami, 2016-02-07:
Yup you're right there is a conflict with python. But not quite sure how to resolve it. 

- I tried to `import vtk` but crashed python on both occasions `directorPython` & `python`
- I reinstalled vtk5 with qt using `brew uninstall vtk5 && brew install vtk5 --with-qt` but got the same error

Here is the output of `otool -L ./build/lib/*dylib | grep Python` 
[http://pastebin.com/QkKhiHWQ](url)

It all started during the build of drake where i had problems with PythonQt and QT. I have Qt5 and cmake needs Qt 4xx. I thought ` install vtk5 --with-qt` would solve the problem. 
is there a direct way to resolve this conflict, or should I do a clean build for drake?

zamzami, 2016-02-07:
drake-visualizer worked from the precompiled release. Is there a way to link this visualizer to drake built from source? (Naïve copy and replacing the bin folder doesn't work :D )

patmarion, 2016-02-07:
you could try a symlink of just build/bin/drake-visualizer.

For your Qt issue, most cmake projects determine which version of Qt to use by looking for the qmake program.  If you run "which qmake" then you'll see which qmake is in your PATH, and if you run "qmake -v" you'll see the version.  If you have qmake 5 in your PATH then your environment probably isn't compatible for a successful build of the director, because it needs qt4.

On the python issue. if you compiled vtk yourself, it probably used the system python and not homebrew python.  When you brew install vtk5 it is careful to select the right version of python.  You should use homebrew to install vtk and uninstall any version of vtk that you may have installed elsewhere.  See my post here for an alternate method to homebrew install vtk without compiling:

https://github.com/RobotLocomotion/drake/issues/1673#issuecomment-180042814

zamzami, 2016-02-08:
drake-visualizer symlink solution worked per se, but it doesn't work when `BotVisualizer` calls it from matlab. 
This is the output of a successful drake-visualizer symlinked session opened outside matlab:
`Last login: Mon Feb  8 01:07:24 on ttys003
ziads-MBP:~ ZZ$ /Users/ZZ/Downloads/drake-v0.9.11-mac/build/bin/drake-visualizer ; exit;
2016-02-08 01:17:21.569 drake-visualizer[51421:214347] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead. 
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
`

while running contactImplicitBrick still gives me: 
`Error using BotVisualizer (line 48)
Failed to automatically start up a viewer (or to receive the ack, see https://github.com/RobotLocomotion/drake/issues/317)`

I have already solved the qt issue and `qmake -v` outputs `QMake version 2.01a
Using Qt version 4.8.7 in /usr/local/Cellar/qt/4.8.7_2/lib`

I never compiled vtk outside homebrew. I tried to uninstall and install your dropbox version and then made a clean build for director but got the following error 
`In file included from /usr/local/opt/vtk5/include/vtk-5.10/vtkPythonUtil.h:21:
/usr/local/opt/vtk5/include/vtk-5.10/vtkPython.h:85:10: fatal error: 'Python.h'
      file not found
#include <Python.h>`
Complete build report is here [http://pastebin.com/WNEhxxRt](url)
I'm sorry for the persistent frustrating errors, but I never ran into any installation problem while building drake v0.9.8 before the change in code architecture. Thank you for your patience @patmarion :)

RussTedrake, 2016-02-08:
If the visualizer is open before you call BotVisualizer, it should use the one that is open instead of trying to launch the original.  (you can find the simple string match that it tries on the command line argument to make sure the name of your symlink will match).

patmarion, 2016-02-08:
Are you sure you've run drake's install_prereqs.sh script, and that is has run to completion without errors?  The error message saying Python.h is not found suggests that you do not have homebrew python installed, which should occur when you run install_prereqs.sh.

zamzami, 2016-02-08:
Thx @RussTedrake that have worked. I should open the visualizer first. 
for @patmarion i actually ran install_prereqs.sh dozen of times :) here is the output [http://pastebin.com/R7K85CrV](url)

zamzami, 2016-02-08:
I'll close this as the suggested turnaround solved the issue. Though I really hoped to solve the root cause issue.

closed 2016-02-08 by zamzami

patmarion, 2016-02-08:
Your build log references /Library/Frameworks/Python.framework.  That seems to be breaking your build.  Did you install this manually?  Where did this come from?  Apple's system version of Python is located in /System/Library/Frameworks/Python.framework and Homebrew's Python is /usr/local/Frameworks/Python.framework.  Your build will be successful when it uses Homebrew's python.  Run this command:

  $ which python

If it doesn't say /usr/local/bin/python then you will have a problem.  The build decides which Python to use by looking for the python program is in your PATH.  After you have resolved this issue, so that Homebrew Python is in your path, then you need to clean and remake Director:

cd externals/director
make clean && make drake-distro

zamzami, 2016-02-11:
Thanks for the pedagogic explanation @patmarion I appreciate it :). After consulting `brew doctor` and  removing /System/Library/Frameworks/Python.framework the system still referred to `/usr/bin/python` so I had to change `/etc/paths` so that `/usr/local/bin` is prioritized which is not the case by default. 
Now Director works like a charm and Drake Visualizer is just beautiful :+1: