RobotLocomotion/drake #1697: make -j in drake-distro broken

tkoolen, 2016-02-08:
It appears that running make -j in drake-distro with all supported externals is broken (at least on OS X). Just running make is fine. Not sure, but I think it's caused by the textbook because of this line in the build output:
```
make[3]: *** [textbook-prefix/src/textbook-stamp/textbook-build] Error 2
```
Any ideas.

psiorx, 2016-02-08:
Not sure if this is related but I had to run `sudo pip install beautifulsoup4` to get the textbook stuff to go through.

rdeits, 2016-02-08:
`pip install beautifulsoup4` is included in the textbook repo's `install_prereqs.sh`, so you should have that already.

tkoolen, 2016-02-09:
Yup, I have beautifulsoup4 installed already. Still broken.

tkoolen, 2016-02-09:
I think something is going wrong during the download step. I don't get any errors, but my drake-distro/drake/doc/textbook directory is empty.

mwoehlke-kitware, 2016-02-24:
What do I need to reproduce this?

tkoolen, 2016-02-24:
build on OSX, `make options` in drake-distro, select WITH_ALL_SUPPORTED_EXTERNALS, `make -j`.

RussTedrake, 2016-02-27:
confirmed that this is resolved.  thanks @mwoehlke-kitware

closed 2016-02-27 by RussTedrake