RobotLocomotion/drake #1701: drakeOptimization and drakeLCMSystem are not installed with pods_install and -fPIC

billhoffman, 2016-02-09:
In working on the CI Windows build, I noticed that I needed to add drake/drake/pod-build/lib/Release to the path instead of just drake/build/lib where the rest of the dlls are.  I tracked it down to some missing install commands.  With the following change I can get by with just these in the path:
${workspace}/drake/build/bin
${workspace}/drake/build/lib

Is there a reason these are not installed?
(also noticed this:
set_target_properties(drakeOptimization PROPERTIES COMPILE_FLAGS -fPIC) 
which will only work with the gcc compiler.

```
$ git diff
diff --git a/drake/solvers/CMakeLists.txt b/drake/solvers/CMakeLists.txt
index a27563c..bfbd729 100644
--- a/drake/solvers/CMakeLists.txt
+++ b/drake/solvers/CMakeLists.txt
@@ -40,6 +40,7 @@ else()
 endif(snopt_c_FOUND)
 add_library_with_exports(LIB_NAME drakeOptimization SOURCE_FILES ${optimization_files})
 set_target_properties(drakeOptimization PROPERTIES COMPILE_FLAGS -fPIC)
+pods_install_libraries(drakeOptimization)
 if (snopt_c_FOUND)
   pods_use_pkg_config_packages(drakeOptimization snopt_c)
 endif(snopt_c_FOUND)
diff --git a/drake/systems/CMakeLists.txt b/drake/systems/CMakeLists.txt
index c30810e..0d49e75 100644
--- a/drake/systems/CMakeLists.txt
+++ b/drake/systems/CMakeLists.txt
@@ -7,7 +7,7 @@ if (LCM_FOUND)
   add_library_with_exports(LIB_NAME drakeLCMSystem SOURCE_FILES LCMSystem.cpp)
   add_dependencies(drakeLCMSystem drake_lcmtypes lcmtype_agg_hpp)
   pods_use_pkg_config_packages(drakeLCMSystem lcm)
-
+  pods_install_libraries(drakeLCMSystem)
   drake_install_headers(LCMSystem.h)
 endif()
 

```

RussTedrake, 2016-02-10:
Just an oversight on my part — those were added recently.  Thanks for adding the installs.

And cleaning up the fPIC would be good.  I know it’s there, but only generates a warning on the other compilers.  Much better to clean it up.

mwoehlke-kitware, 2016-02-23:
The easy way to deal with PIC is with `CMAKE_POSITION_INDEPENDENT_CODE`... how do we feel about bumping the minimum required CMake to 2.8.10?

jamiesnape, 2016-02-23:
Could cause issues for the build servers. The cloud based ones are mostly ok (Jenkins runs 3.4.3, Appveyor 3.4.0, not sure about Travis CI), but the others would may need upgrading. Check with @RussTedrake.

RussTedrake, 2016-02-23:
The reason for our current minimum version is because of the default packaging on ubuntu.  I don't want to lengthen the list of installation instructions (eg with a new ppa) unless it's truly warranted.   But if you guys are going to be bumping into these issues frequently then i'd definitely consider it

iamwolf, 2016-02-23:
Apologies for jumping into this discussion; Trusty, the oldest release that is presently [still supported](https://wiki.ubuntu.com/LTS), provides [cmake version 2.8.12](http://packages.ubuntu.com/trusty/cmake) without ppa. This is then also available on Travis if trusty is enabled (``dist: trusty``), cf. [here](https://docs.travis-ci.com/user/trusty-ci-environment/).

rdeits, 2016-02-23:
Bumping Travis up to Trusty requires us to resolve #1553, which is not a huge problem.

PR #1742 (2016-02-24): Add missing install rules
Add some missing install rules for drakeOptimization and drakeLCMSystem.

This "mostly" fixes #1701; however, I didn't do anything about the PIC issues as I would prefer to address this with CMake support (as discussed in #1701).
closed 2016-02-24 by RussTedrake