RobotLocomotion/drake #1702: runAtlasDynamics fails on drake001

tkoolen, 2016-02-10:
See e.g. the test result in this build of master: http://kobol.csail.mit.edu/cd/testDetails.php?test=3616246&build=21232. Path LCP reaches its iteration limit.

Possibly related: I see this warning in the test output, which doesn't appear on my system:
```
[ Warning: This timestep violated our assumptions about which contacts could
possibly become active in one timestep.  Consider reducing your dt.  If it
seems to happen a lot, then please let us know about it.] 
```

tkoolen, 2016-02-10:
@billhoffman, for this type of issue, it would be very useful to see when this particular test first started failing on drake001. Is there any way to see that quickly with CDash?

RussTedrake, 2016-05-07:
not sure why drake001 would be any different?

tkoolen, 2016-09-20:
Closing since drake001 is not a thing anymore. Please open a different issue if this happens on a different machine.

closed 2016-09-20 by tkoolen