RobotLocomotion/drake #1703: Appveyor

jhoare, 2016-02-10:
Appveyor builds are often failing while attempting to download installation packages from dropbox. 

Unclear if this is a problem with Appveyor, or DropBox, or some combination of the two. 

```
Build started
git clone -q --depth=5 https://github.com/RobotLocomotion/drake.git C:\projects\drake
git fetch -q origin +refs/pull/1698/merge:
git checkout -qf FETCH_HEAD
Running Install scripts
(new-object net.webclient).DownloadFile('https://dl.dropboxusercontent.com/u/3432353/pkg-config.exe', 'c:\pkg-config.exe')
Exception calling "DownloadFile" with "2" argument(s): "Unable to connect to the remote server"
At line:1 char:1
+ (new-object net.webclient).DownloadFile('https://dl.dropboxusercontent.com/u/343 ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : WebException
 
Command executed with exception: Exception calling "DownloadFile" with "2" argument(s): "Unable to connect to the remote server"
```

RussTedrake, 2016-02-11:
we decided that we should move the pkg-config-lite out of dropbox and put it in an external (like, or even with, the windows gtk external).

jamiesnape, 2016-02-23:
I updated the link to the S3 bucket that we use for provisioning Jenkins for now, and so far that seems to be working out.

RussTedrake, 2016-02-23:
Will close (csn reopen if we see it again).

closed 2016-02-23 by RussTedrake