RobotLocomotion/drake #1705: TrigPoly in a eval call with square roots and if cases

tarquasso, 2016-02-11:
I am trying to run this:

```
obj = PlanarRigidBodyManipulator('tensionWParams.urdf');
nq = obj.getNumPositions;

q=msspoly('q',nq);
s=msspoly('s',nq);
c=msspoly('c',nq);
qt=TrigPoly(q,s,c);
p=obj.getParamFrame.getPoly;
pobj = setParams(obj,p);
[H,C,B] = manipulatorDynamics(pobj,qt,qd);
parametricPositionConstraint = pobj.position_constraints{1}.eval(qt); % calls drakeFunction.kinematic.CableLength/eval()

err = [err;parametricPositionConstraint];

```
when executing 
```
parametricPositionConstraint = pobj.position_constraints{1}.eval(qt)
````
I get this error:
```
Undefined function 'sqrt' for input arguments of type 'TrigPoly'.

Error in drakeFunction.kinematic.CableLength/eval (line 48)
C = sqrt(vec'*vec); % abs distance between the two pulleys
```

qt is a TrigPoly object. I am trying to perform a square root operation on a TrigPoly object, because I need that for calculating C, the absolute distance between the pulleys. I guess the square root is not implemented since the result of the square root operation wouldn't be any more polynomial... 

Obviously, I could do
Csq = vec'*vec;
and then see if the algebra following this can be rewritten to use Csq instead. But that is not going to serve me any good either since there are a bunch more 'if' cases and operations on C following that:

```
...
            cvec = vec/C; 
            if alignment>0 
              s = (r2-r1)/C;  
              alpha = asin(s);
              pt1 = last_pt + r1*axis2rotmat([obj.pulley(i-1).axis;-pi/2-alpha])*cvec;
              pt2 = pt + r2*axis2rotmat([obj.pulley(i).axis;-pi/2-alpha])*cvec;
              ....
```
I think the solution is?:
I need to populate these additional lines of the matrix "err" based on the measured data. There won't be any hope to do it symbolically...?

RussTedrake, 2016-02-11:
If the system is not trig / poly than it’s not going to work.  agreed that the simplest solution would be writing everything below in terms of Csq

we don’t actually need it to be trig poly to extract the parameters… we were just reusing the tools for convenience (because they supported the lumped parameters).

closed 2016-02-12 by RussTedrake

tarquasso, 2016-02-12:
I tried writing everything in form of Csq, but there are a) a few if 
cases, b) several divisions by Csq, c)and asin() operations on something 
constructed by Csq. So I figured that is not a viable route to take in 
this case. Even if I do not use TrigPoly, the problem is the same just 
for a more generic msspoly.

Instead, I hope the solution lies in writing a nonlinear program that 
takes q(2:3) and qd(2:3) as input, uses the cablelength, dcablelength 
and ddcablelength for the cost function and then calculates q(1) and 
qd(1). But this would open up a new topic now anyways.