RobotLocomotion/drake #1708: How do we know which joints have actuators in an SDF file?

RussTedrake, 2016-02-12:
We used transmission elements in urdf.  How does one specify actuators in sdf?  Given what I see here:
http://gazebosim.org/tutorials?tut=ros_control

and here: 
http://answers.gazebosim.org/question/321/equivalent-transmission-or-actuator-tags-in-sdf-compared-to-urdf/

it’s hard to know.  But there is an effort (limit) that can be assigned to each joint.  So what does gazebo do?  Does it allow a torque to be applied at every joint (unless the effort limit is set to zero)?

needed by #1653

mauricefallon, 2016-02-13:
IHMC use sdf. this is val in sdf
https://bitbucket.org/ihmcrobotics/ihmc-open-robotics-software/src/10305dc5c88393bc5017f6493887ed41bf53650c/Valkyrie/resources/models/val_description/sdf/valkyrie_sim.sdf?at=develop&fileviewer=file-view-default

    <joint name='leftAnklePitch' type='revolute'>
      <child>leftAnklePitchLink</child>
      <parent>leftKneePitchLink</parent>
      <axis>
        <xyz>0 1 0</xyz>
        <limit>
          <lower>-0.8644</lower>
          <upper>0.875</upper>
          <effort>205</effort>
          <velocity>11</velocity>
        </limit>
        <dynamics>
          <damping>0.1</damping>
          <friction>0</friction>
          <spring_reference>0</spring_reference>
          <spring_stiffness>0</spring_stiffness>
        </dynamics>
        <use_parent_model_frame>1</use_parent_model_frame>
      </axis>
      <physics>
        <provide_feedback>1</provide_feedback>
        <ode>
          <provide_feedback>1</provide_feedback>
          <implicit_spring_damper>1</implicit_spring_damper>
          <cfm_damping>1</cfm_damping>
          <limit>
            <cfm>0</cfm>
            <erp>0.2</erp>
          </limit>
        </ode>
      </physics>
    </joint>

There's a gazebo command line tool for conversion from urdf->sdf. Biggest issue is that pdf collapses all fixed links down to a single inertia.

Recently Gazebo posted on their wiki about planning to 'merge sdf and urdf'

RussTedrake, 2016-02-13:
Right.  Thanks @mauricefallon .  In drake, I want to populate the input coordinates with only the forces/torques that you can actually apply to the robot.  I've also confirmed with OSRF that they don't have actuator/transmission tags specifically for sdf.  For instance, in our Acrobot.sdf, I'd like to specify that the elbow has an actuator, but the shoulder does not.

So I think we're left with a decision:
A) we could say that every single-axis joint has an actuator attached, unless the effort limit is set to zero.
B) we could say that every single-axis joint which specifies a [gearbox_ratio](http://sdformat.org/spec?ver=1.6&elem=joint#joint_gearbox_ratio) has an actuator attached.  

Given the val model, perhaps we have to go with A?

mauricefallon, 2016-02-14:
I might be incorrect but from my VRC recollections, but I think that an SDF has an include tag that can include a urdf of a robot directly. Thus this might allow you to just support URDF for this.

RussTedrake, 2016-02-14:
There was actually a pull request that would have enabled it, but it got killed
(merge was declined). The thinking was that you didn’t want everyone who
implements an sdf parser to also have to implement a urdf parser.

[https://github.com/RobotLocomotion/drake/issues/1708#issuecomment-183959439] .[https://github.com/notifications/beacon/AGJNNJzeRoJqX7pihB1viEvx-Andgtb-ks5pkMy1gaJpZM4HY9kw.gif]

hsu, 2016-02-16:
Some draft personal notes [here](https://bitbucket.org/osrf/gazebo_design/src/090bd9e7997523a97002ca55b2ac253b612845d6/transmission/transmission.md?fileviewer=file-view-default), inputs welcome.

hsu, 2016-02-16:
@RussTedrake are you referring to [this pull request](https://bitbucket.org/osrf/sdformat/pull-requests/165/urdf-format-support-in-sdf/diff) that tries to define URDF in SDF?

RussTedrake, 2016-02-16:
that's the one.  thanks for contributing here.

RussTedrake, 2016-02-16:
also from @hsu :
As you mentioned in your discussion thread, currently any plugin can apply force/torque on any joint by calling Joint::SetForce, efforts applied is limited by velocity and effort limits specified on a joint.

scpeters, 2016-02-17:
I'm a little late to this thread, but I'll try to catch up:

We should add a transmission and/or actuator concept to sdformat. @hsu listed some of our preliminary discussion on that.

@mauricefallon the lumping of bodies with fixed joints is now optional, as of [sdformat pull request 133](https://bitbucket.org/osrf/sdformat/pull-requests/133/added-urdf-gazebo-extension-option-to/diff#Ltest/integration/fixed_joint_reduction_disabled.urdfT121), try adding something like the following to disable it:

~~~
<gazebo reference="joint12a">
  <disableFixedJointLumping>true</disableFixedJointLumping>
</gazebo>
~~~

The default value of the `<effort>` tag is `-1`, which corresponds to an unlimited value. Setting it to `0` seems like a reasonable proxy for an un-actuated joint until we have a better definition.

There's also been a request for asymmetric effort limits ([drcsim issue 361](https://bitbucket.org/osrf/drcsim/issues/361)).

closed 2016-02-20 by RussTedrake

aespielberg, 2018-02-24:
This issue seems to persist in the current version of drake.  Removing the effort limits tags, or manually setting the value to -1.0, the sdf parser does not add an actuator.  I have taken the double pendulum sdf, removed the effort tag from one of the joints, and parsed it to a plant.  Calling `plant->get_rigid_body_tree().get_num_actuators();` then returns 0.  This seems to be a bug.

commit 67747606 (2016-02-20): implements basic actuator logic
resolves #1708