RobotLocomotion/drake #1712: collisionRaycast gets returns from the margin world even when use_margins=false

RussTedrake, 2016-02-15:
Added a super simple test (lidarTest.cpp) in my pull request #1711.  the collision world is a set of boxes, with a lidar sim.  it starts getting returns exactly at 0.05 before the box geometry should start despite (i think) making the raycast call with use_margins=false (the default).

@gizatt -- can you take a look?

RussTedrake, 2016-02-15:
hmm..  adding the following line:

```
            double margin = ray_callback.m_collisionObject->getCollisionShape()->getMargin();
```

to line 577 of BulletModel.cpp confirms that I am calling it with no_margins, and the shape that registers the hit has no small_margin set.  So I don’t understand why we have the pre-mature hit?

RussTedrake, 2016-02-15:
separately, putting a breakpoint on BulletModel.cpp and stepping through the world position of the returns (the end_eigen variable) reveals that the y coordinate bounces all over, from 0.995 to essentially 1.  Why such poor resolution?

RussTedrake, 2016-02-15:
@gizatt - I’m convinced this is not a simple use_margin logic bug.  removing the assignment to you.  :)

RussTedrake, 2016-02-15:
Just to post some numerics on the craziness… the following scans are supposed to walk along a box at y=-1, to x=0, then y=+1.  These are the returns I get (in world coordiantes).  This is with use_margins = false.  Note the stretch where I get accuracy to 1e-9 (which is the small_margin that we set, e.g. the expected behavior).  But the rest are much closer to the 0.05, or large_margin.  Something is very fishy here.

```
x: -1.00939, y: -1.00062
x: -1.00406, y: -1.00012
x: -0.998739, y: -1
x: -0.993419, y: -1
x: -0.9881, y: -1
x: -0.982781, y: -1
x: -0.977462, y: -1
x: -0.972143, y: -1
x: -0.966824, y: -1
x: -0.961504, y: -1
x: -0.956183, y: -1
x: -0.950861, y: -1
x: -0.945537, y: -1
x: -0.94021, y: -1
x: -0.934882, y: -1
x: -0.929551, y: -1
x: -0.924218, y: -1
x: -0.918881, y: -1
x: -0.91354, y: -1
x: -0.908196, y: -1
x: -0.902848, y: -1
x: -0.897509, y: -0.999865
x: -0.892255, y: -0.998922
x: -0.887004, y: -0.997989
x: -0.881757, y: -0.997065
x: -0.876513, y: -0.996151
x: -0.871271, y: -0.995245
x: -0.865271, y: -1
x: -0.859879, y: -1
x: -0.85448, y: -1
x: -0.849074, y: -1
x: -0.843661, y: -1
x: -0.838239, y: -1
x: -0.83281, y: -1
x: -0.827372, y: -1
x: -0.821925, y: -1
x: -0.816469, y: -1
x: -0.811004, y: -1
x: -0.805529, y: -1
x: -0.800043, y: -1
x: -0.795474, y: -0.995489
x: -0.789811, y: -0.996351
x: -0.784128, y: -0.997208
x: -0.778425, y: -0.99806
x: -0.772702, y: -0.998907
x: -0.766958, y: -0.99975
x: -0.761334, y: -1
x: -0.755756, y: -1
x: -0.750164, y: -1
x: -0.744558, y: -1
x: -0.738939, y: -1
x: -0.733305, y: -1
x: -0.727657, y: -1
x: -0.721993, y: -1
x: -0.716314, y: -1
x: -0.710619, y: -1
x: -0.704908, y: -1
x: -0.69918, y: -1
x: -0.693436, y: -1
x: -0.687673, y: -1
x: -0.681893, y: -1
x: -0.676095, y: -1
x: -0.670278, y: -1
x: -0.664443, y: -1
x: -0.658587, y: -1
x: -0.652712, y: -1
x: -0.646817, y: -1
[... 519 lines cut ...]
x: -0.887004, y: 0.997989
x: -0.892255, y: 0.998922
x: -0.897509, y: 0.999865
x: -0.902848, y: 1
x: -0.908196, y: 1
x: -0.91354, y: 1
x: -0.918881, y: 1
x: -0.924218, y: 1
x: -0.929551, y: 1
x: -0.934882, y: 1
x: -0.94021, y: 1
x: -0.945537, y: 1
x: -0.950861, y: 1
x: -0.956183, y: 1
x: -0.961504, y: 1
x: -0.966824, y: 1
x: -0.972143, y: 1
x: -0.977462, y: 1
x: -0.982781, y: 1
x: -0.9881, y: 1
x: -0.993419, y: 1
x: -0.998739, y: 1
x: -1.00406, y: 1.00012
x: -1.00939, y: 1.00062
```

avalenzu, 2016-02-15:
Can you post a visualization of what this looks like? Your description
reminds me of the ripples in the ray cast height maps we encountered before
the DRC.

RussTedrake, 2016-02-15:
It's a bit hard to see at these scales, but the laser is at the red star, and the returns are oddly symmetric:
[image]

and here is a zoom in on the bottom line.... 
[image]

RussTedrake, 2016-02-15:
oops.  lost the red star.  lidar (raycast origin) is at (-1,0)

amcastro-tri, 2016-05-07:
Very relevant to what I am working on right now. Thank you @RussTedrake for driving my attention to this. I'll take a look.

amcastro-tri, 2016-05-12:
Here is a close up that I just made. The plot below shows distance as a function of angle. I run the same case but with a much smaller yaw range (-0.1 to 0.1) and I used 201 points (an odd number so I could see the zero angle case). The case is actually repeatable with different number of yaw points so I can run it with 5 points instead for easy debug. What the 201 points case shows is that there is a distinct point at which the plot breaks (see the data tips in the plot below). Break points are independent of range resolution.

At exactly zero angle the lidar return is correct. 
[image]

amcastro-tri, 2016-05-12:
The ray test happens deep inside bullet in `btCollisionWorld.cpp:917` with the call to `btCollisionWorld::rayTestSingle` (in `btCollisionWorld.cpp:262`). 
I simplified the problem to a single ray and I still get the wrong value. It looks like it is a bug within Bullet. I will write a very small bullet demo tomorrow outside drake with the same conditions to see if the problem persists.

amcastro-tri, 2016-05-12:
Even deeper, Bullet is doing the final raycast with `btSubsimplexConvexCast::calcTimeOfImpact` at `btSubSimplexConvexCast.cpp:38`. There is some sort of iteration going on that I do not understand but clearly an overkill for ray-to-box test.

sherm1, 2016-05-12:
Alejandro, if you can get this down to a simple demonstration of a Bullet problem we can get Erwin to take a look at it and probably get a quick fix. Let me know when you get to a point where you want him involved and I will attempt to pull him in.

amcastro-tri, 2016-05-12:
After 8 (maxIter=56, it starts counting from 64) iterations in `btSubsimplexConvexCast::calcTimeOfImpact` the method returns with `dist2=8.3221703372389529e-05` which is smaller than the required `epsilon=0.0001`.
As far as I know bullet has [3 implementations of ray/convex casting](http://www.continuousphysics.com/Bullet/BulletFull/classbtConvexCast.html). Maybe we could tray using another implementation rather than [btSubsimplexConvexCast](http://www.continuousphysics.com/Bullet/BulletFull/classbtSubsimplexConvexCast.html)?

amcastro-tri, 2016-05-12:
@sherm1: that'd be really great! I work tomorrow in that little demo. From Drake I am working with only one box and one ray and I still have the problem. So I think it should be possible to write a simple Bullet test code that then Erwin can work with

amcastro-tri, 2016-05-14:
@sherm1: I was working on the simple demo for Erwin and I found  a solution to the problem (#2354). However I'll finish it up (I just have to clean it up now)because it'd be nice to have his input on this. Specially see if it'd be convenient to have the option to externally set this tolerance constant now buried deep inside Bullet.

liangfok, 2016-05-17:
I'm going to close this issue since https://github.com/RobotLocomotion/drake/pull/2354 was merged. Please re-open if this issue is not 100% resolved.

closed 2016-05-17 by liangfok