RobotLocomotion/drake #1713: fourbar.sdf does not match fourbar.urdf

RussTedrake, 2016-02-15:
Currently lives only in #1711.  Once this works, we should add_test to call the comparison.
The reason is because the fourbar needs to be "closed" in the zero-configuration order for the sdf to be fully specified.   That's how sdf gets around having multiple transforms specified for the joints (and avoids calling resolveConstraints). 

I'm ok with making a simpler four-bar system (so the closed configuration has angles at pi/2 type intervals).  perhaps we'd call that FourBar3.urdf and move FourBar.sdf to FourBar3.sdf.  

While not urgent, I definitely think we want to have this to test the sdf/urdf parsing.

RussTedrake, 2016-05-07:
@liangfok might have resolved this by now.

tri-ltyyu, 2019-03-22:
@RussTedrake, please close if this is not important anymore

RussTedrake, 2019-03-22:
i guess it's still relevant as long as the `examples/simple_fourbar` directory is sticking around.  might be relevant as @amcastro-tri implements loop joints?   otherwise perhaps we should zap the examples?

jwnimmer-tri, 2021-11-02:
@joemasterjohn how about we just purge the files?  We can always resurrect them as part of #9566, if we ever decide to do that.

nyxaria, 2021-11-24:
Is there a python example for the 4bar linkage? How would I go about porting this to python if not?

rpoyner-tri, 2022-04-05:
I foresee no forward path for URDF `loop_joint` in Drake, and would happily just delete the FourBar*.urdf files. However, they are in the set of files installed in binary releases. Is there some mechanism/ritual for deprecation, or do we just delete them and move on?

jwnimmer-tri, 2022-04-11:
See https://drake.mit.edu/stable.html#model-files:

> Note in particular that any data files provided by Drake (e.g., SDFormat or URDF models) are not incorporated into the “Stable API”, even though some of those files are distributed in our stable releases for demonstration purposes. We may alter the models (e.g., changing the collision geometry or joint limits) or remove a model entirely, without prior notice.

Thus, you can remove them immediately if you like.

Or, the process for deprecating them (if you want to) is as follows:

(1) Open a pull request that adds a README to their folder explaining the deprecation, with 2022-MM-01 in the README text.  Label it as "release notes: newly deprecated" so that the deprecation notice makes it into the next release notes.

(2) Once the date is reached, we remove the files and the README.

PR #16951 (2022-04-16): [examples] Remove obsolete examples built on loop_joint
Closes: #1713

Per discussion in the issue, data files do not require deprecation.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16951)
commit a74e38e0 (2022-04-16): [examples] Remove obsolete examples built on loop_joint (#16951)

* [examples] Remove obsolete examples

Closes: #1713

Neither the URDF nor the SDFormat four-bar examples can any longer be
run by drake, since there is no support for closed kinematic loops.

Per discussion in the issue, data files do not require deprecation.
closed 2022-04-16 by rpoyner-tri