RobotLocomotion/drake #1715: try enabling lcm for 32-bit windows

RussTedrake, 2016-02-15:
@jhoare -- you said that lcm was working now on 32-bit windows (or specifically, that you didn't need the if 32-bit logic in the lcm CMakeLists.txt).  But lcm is not being build by the drake test suite on 32-bit because of the logic in the drake superbuild at https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L38 .

If you pull the lcm option ON out of those conditionals and get it to build on win32 (really 32), then fantastic.  otherwise, we should probably put the win32 check back into the lcm CMakeLists.txt, too?  The best, of course, would be to successfully build lcm on win32.

billhoffman, 2016-02-15:
It will not work with the super build and 32 bit because the gtk libraries are 64 bit.  So, the check needs to stay.

RussTedrake, 2016-02-15:
Good point.  Does lcm fail with a useful error on 32-bit because it cannot find glib?  if yes, then we can simply close this.  If no, then we should restore that original check (assuming I remember correctly that it was removed)

jhoare, 2016-02-26:
It looks like we should be able to build lcm on windows32 and windows 64. There was an issue with gtk-pod which always installed 64bit version of gtk. 

See https://github.com/RobotLocomotion/gtk-pod/pull/3

RussTedrake, 2016-02-26:
Awesome. Good catch. I’ve merged the other PR. Please give it a shot!

PR #1760 (2016-02-29): Enabled LCM build on Windows 32bit.
Fixes #1715
closed 2016-02-29 by RussTedrake