RobotLocomotion/drake #1718: find command (used to find package:// paths) does not work on windows

RussTedrake, 2016-02-16:
in xmlUtil.cpp we do
```
    istringstream p(exec("find -L "+token+" -iname package.xml"));
```
which does not work on windows.
```
dir /b/s 
```
could work on the standard shell, but does not work in the git bash shell, etc.  we need a more robust solution.
boost::filesystem could do it, but we've been avoiding boost.
stl has a filesystem implementation (see http://en.cppreference.com/w/cpp/experimental/fs/recursive_directory_iterator and http://www.technical-recipes.com/2014/using-boostfilesystem/#RecursiveSearch), but it appears to not be available yet on mac clang at least.

RussTedrake, 2016-02-16:
btw - the result is that many of the unit tests on windows spew
```
9: FIND: Parameter format not correct
9: FIND: Invalid switch
9: FIND: Parameter format not correct
9: FIND: Invalid switch
9: FIND: Parameter format not correct
9: FIND: Invalid switch
9: FIND: Parameter format not correct
9: FIND: Invalid switch
9: FIND: Parameter format not correct
9: FIND: Invalid switch
```

RussTedrake, 2016-03-02:
this is blocking @sherm1, and anyone trying to visualize the car sim on windows.

billhoffman, 2016-03-02:
I had seen that but the tests still pass for some reason...    I guess this should be a cmake script instead.

sherm1, 2016-03-02:
I noticed two possible platform dependencies: 
- behavior of the `find` command as mentioned above (assumes unix)
- separator character in `path` (assumes `:`)

The separator is conventionally `;` on Windows but it isn't obvious to me where the `searchDirectory()` method is getting its path from. If it is being constructed internally then it might be using `:` even when running on Windows.

RussTedrake, 2016-03-02:
The sims run, but the visualizer (which does not get tested via ctest) won't load the images.

I guess we could potentially replace it with a cmake -E call.  That's a bit crazy, but could work :)

billhoffman, 2016-03-02:
Maybe something like this: 

p(exec("cmake -Dname=" + token + "package.xml -P find.cmake"));

What exactly is it trying to do?  Maybe if I understood that.  Another option is to grab the Directory class from CMake/VTK/ParaView from kwsys.  It is a c++ directory walking class.

RussTedrake, 2016-03-03:
A c++ directory walking class is much more inline with what we want. We’re just
trying to locate directories in the search path that contain a file that matches
this expression.
I’ve been very happy with spruce as a super lightweight platform independent
implementation (
https://github.com/RobotLocomotion/drake/blob/master/drake/thirdParty/spruce/include/spruce.hh
). The only thing it’s missing is the ability to list a directory. Maybe adding
that one call to spruce might be the cleanest solution?

sherm1, 2016-03-03:
> Maybe adding that one call to spruce might be the cleanest solution?

:+1:

commit 3f85e4d3 (2016-03-24): Stop running the find command to traverse directory trees.  (Fixes #1718)

This change pulls in the "tinydir" library
https://github.com/cxong/tinydir -- It's BSD licensed and cross-platform.
commit 8e6f5954 (2016-03-24): Merge pull request #1873 from sammy-tri/issue_1718

Resolve #1718 by no longer calling exec for the find command
closed 2016-03-24 by RussTedrake