RobotLocomotion/drake #1720: build bullet on windows

RussTedrake, 2016-02-16:
bullet is currently disabled on windows.  this was not previously such a big deal, because all of the examples requiring bullet also required an lcp solver (which we also don’t have on windows yet).  but now the car simulation, and the lidar sims could work on windows if we have just bullet.  

bullet should build on windows, and i was pretty close to getting it to work the last time i tried.

jonathan-owens, 2016-02-23:
It seems to build fine on my Windows 10 VM with both premake + Visual Studio and CMake + Visual Studio.

billhoffman, 2016-02-23:
@jonathan-owens can you create a PR with this enabled by default and we can see if it passes the CI windows machines.

RussTedrake, 2016-02-24:
it’s possible that we’ve resolved the issues since my last attempt by simply
increasing the minimum required version of msvc. can you try enabling it and let
the build servers take a shot at it?

jonathan-owens, 2016-02-24:
I just made a pull request with bullet on by default, waiting for the builds to run. Forgot to mention this issue in the PR

billhoffman, 2016-03-01:
Fixed here: https://github.com/RobotLocomotion/drake/pull/1746

closed 2016-03-01 by billhoffman