RobotLocomotion/drake #1723: Many type conversion warnings on Windows

jamiesnape, 2016-02-17:
Mostly related to lcm/lcm_coretypes.h header, e.g. in systems\LCMSystem.cpp. See http://kobol.csail.mit.edu/cd/viewBuildError.php?type=1&buildid=21820.

sherm1, 2016-03-02:
Building Win64 from scratch on the command line I got ~1300 warnings.

jamiesnape, 2016-03-02:
For the record, we stop counting warnings after 50 on CDash.

david-german-tri, 2016-05-16:
Problem still exists.  Updated cdash link: https://drake-cdash.csail.mit.edu/viewBuildError.php?type=1&buildid=20298

jwnimmer-tri, 2016-05-16:
FWIW, for things like this, I recommend fixing the primary platforms first (gcc and clang) by turning on the analogous warnings there, and fixing the code -- and only then looking at Windows.  The closer we can keep the Ubuntu warnings and the Windows warnings, the better.

david-german-tri, 2016-05-16:
Good idea for the subset of warnings from Drake, but most of them are from an external.

billhoffman, 2016-05-16:
Seems like a good place to start would be adding -Wsign-conversion on clang.

sherm1, 2016-05-16:
+1 for fixing this. I filed lcm-proj/lcm#63 on this topic. They enthusiastically requested a PR! I briefly attempted to build lcm standalone and failed miserably on glib as described in the issue.

david-german-tri, 2016-05-16:
`-Wsign-conversion` on Clang produces 4,278 warnings that are neither a subset nor a superset of MSVC `C4267`.  Spot-checking a few, they appear technically-wrong-but-harmless.  I don't think it's worthwhile to fix them en masse.

Honestly, the same seems true about `C4267`, so I've downgraded the whole thing to `priority: backlog`.

sherm1, 2016-05-16:
> Honestly, the same seems true about C4267

I don't agree about C4267. The ones I've checked always involve someone shoving 64 bit values (typically size_t) into 32 bit values (typically int). That is harmless if values are limited to 32 bits, which is typically the case. But disabling that warning means potentially missing legitimate 32-to-64 bit porting problems, which will be harmless just until bit 33 gets set in the field. Leaving the warning enabled causes boy-who-cried-wolf syndrome regarding all the other warnings. I would prefer priorty:low, with gradual repair as files get changed, and by whining to 3rd party code maintainers.

jwnimmer-tri, 2016-07-18:
Just in case anyone comes along and tries to work on this -- I strongly recommend distinguishing between:
(1) warnings during the superbuild (where we should either ignore these reports, or even just disable the warnings entirely), and
(2) warnings in Drake itself (where we should strive to achieve 0 warnings, either by fixing or disabling the current warnings -- and where `-Werror=...` is the best tool to gain and hold progress.

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri