RobotLocomotion/drake #1724: persistent config error on libbot

jscholz, 2016-02-17:
I'm having trouble building with libbot:

```
[jscholz@jscholz-]:~/vc/drake-distro (git: master) $ make
-- Found matlab
CMake Error at CMakeLists.txt:178 (message):
  ----- ERROR: bot2-lcmgl not found
-- Configuring incomplete, errors occurred!
```

This seems related to this issue:
https://github.com/RobotLocomotion/drake/issues/1390

However the problem doesn't go away when I set WITH_LIBBOT to OFF in `make options` (followed by `make configure` and `make`)
https://github.com/RobotLocomotion/drake/issues/1523

Interestingly, libbot seems to build fine from its own directory.  

Is there a fix or workaround for this issue?  I can provide more output if necessary.

Thanks, and looking forward to trying Drake!

mwoehlke-kitware, 2016-02-17:
FYI, this happens when libbot gets built without Java (do you have a JDK installed?). You may need to whack it (libbot) with a hammer to get it to rebuild.

jscholz, 2016-02-18:
worked!

closed 2016-02-22 by RussTedrake