RobotLocomotion/drake #1727: Undefined function or variable 'constructQPDataPointerMex'

alexbuyval, 2016-02-18:
Hi, I am trying to run 'runAtlasWalking' file from /drake-distro/drake/examples/Atlas folder.
However, I have got the following error:

```
Undefined function or variable 'constructQPDataPointerMex'.
Error in atlasControllers.InstantaneousQPController (line 83)
```

although  I have 'constructQPDataPointerMex.cpp' file into drake-distro/drake/systems/controllers folder.
I have got the Matlab 2015b and I have installed the 'Drake' from binary. 
Maybe I missed a important step to compile mex files?

RussTedrake, 2016-02-18:
Hi @alexbuyval .  This mex function depends on having the gurobi solver.  If CMake finds gurobi at configure time, then it will build the mex file.  (Gurobi is a commercial solver, but does grant free academic licenses).

closed 2016-02-18 by RussTedrake

alexbuyval, 2016-02-19:
@RussTedrake  Thank you! Now it works!

oguzhancebe, 2017-08-23:
Hi, I am having the exact same problem, but I have Gurobi at my Matlab path.
I have Matlab 2017a-academic, Gurobi 7.5.1 Academic and Ubuntu 16.04 OS
I added Gurobi to path with subfolders before trying to launch the simulation and I can launch Gurobi matlab examples, by the way.
>which gurobi
/home/oguz/opt/gurobi751/linux64/matlab/gurobi.mexa64

I couldn't see what is causing this issue.

RussTedrake, 2017-08-23:
I'm afraid the mex support of gurobi is not implemented yet in the bleeding edge drake.  We'll have an excellent solution (by wrapping the new c++ classes) soon, but aren't supporting the old code anymore