RobotLocomotion/drake #1731: FLA Simulation Wishlist

psiorx, 2016-02-19:
Here are some of the requirements and nice-to-haves for using Drake as the primary simulator for FLA.

#### Mandatory:
- [x] Quadrotor Dynamics
- [ ] PX4 Software-in-the-loop Integration (send Sensor state and receive control inputs over MAVLink  see Gazebo and/or jMAVSim for open source implementations of this)
- [ ] ROS Integration (Ability to publish robot state as geometry_msgs/PoseStamped, lidar data as sensor_msgs/LaserScan, etc)
- [x] Scanning Lidar
- [ ] IMU
- [ ] Single Point Lidar
- [x] Collisions with ground plane and other static geometry
- [x] Ability to operate in 'headless' mode with no GPU/GUI dependencies.

#### Optional:
- [x] Load SDF robots and environments
- [ ] RGBD Sensor
- [ ] Camera Sensor

I'll continue to update this list as requirements change or new functionality comes online.

RussTedrake, 2016-02-21:
Thanks @psiorx.  These all seem quite doable.  
- I’ve just added #1733 for starting on the ROS compatibility.  Please do make a list of the very specific list of messages you need.  
- Moving the IMU logic over from matlab is simple enough.  
- Is the single point lidar a different sdf tag?  If yes, then just point me to the sdf file and we can parse it.  The existing lidar code should handle that case. 

can you find out a little more about the px4 integration?  does it / can it go through ROS, too?  or if you have the spec we can obviously write the client.  Perhaps you can take that part?

psiorx, 2016-02-21:
I looked into the PX4 integration code as done by jMAVSim and it doesn't seem that bad.  It basically involves sending serialized mavlink IMU messages over UDP on a specific port and listening for the corresponding mavlink actuator commands on another port.  Those can then be translated to LCM to plug directly into Drake.

I'd be happy to take that part if Nick is OK with it.

I'm also willing to bet that the PX4 folks would be happy to add Drake to the list of officially supported simulators if we package up the code and submit a PR.

psiorx, 2016-02-21:
> Is the single point lidar a different sdf tag?

I haven't looked too deeply into the various laser options but I'd imagine it could be handled as a special case of the scanning lidar with min_angle=max_angle and samples set to 1.

psiorx, 2016-02-24:
Made some progress on the mavlink SITL interface today and it seems like the magnetometer readings are required.  I've added it to the wishlist.

tkoolen, 2016-02-27:
> Moving the IMU logic over from matlab is simple enough

In my opinion, the IMU class needs a little bit of rethinking before moving to C++.

* currently, our IMU is perfect. We should at least have thought about how incorporating sensor noise / delay should work
* from a software design point of view, I think it would be nice to first create classes for accelerometers, gyros, magnetometers, and then have an IMU be composed of (a subset of) those.
* right now our IMU returns orientation measurements. There is no physical sensor in an IMU that would directly give you orientation. IMUs that do on-board inertial filtering do return an orientation estimate, so right now our IMU class is an idealization of that. It would be good, at least at some point in the future, to also have a more realistic IMU that does filtering to determine orientation given available measurements. Just something to think about in the design.

RussTedrake, 2016-02-27:
While I agree with all your suggestions, I don’t think we have to perfect the
IMU model before porting to c++. We can port what we have, delete the matlab
implementation (hopefully soon), then improve things in c++. Changing it during
the port could actually make it harder to zap the matlab version.

psiorx, 2016-02-27:
To be more clear about the requirements for the IMU: it should provide 3-axis linear acceleration in body frame, and 3-axis angular rates in body frame (with gaussian additive noise).  I'm not sure if the MATLAB implementation currently does this.

The magnetometer should give the direction of true north modulo magnetic declination in body frame also with gaussian additive noise.  

The orientation is estimated separately in an EKF running within the px4 firmware.

psiorx, 2016-03-25:
@RussTedrake 
This is now working in the FLA integrated system.  One shortcoming, though, is that all of the sensors give their output at every simulation tick.  This is undesirable since the lidar simulation is computationally expensive and does not need to be computed at the same rate as the dynamics.  Because of this, I'm only able to cast around 150 rays while still keeping up with realtime.  This is insufficient for the scan matcher to be robust.

What do you think about supporting a pipeline that would allow sensors to run asynchronously at rates that are independent of the simulation(e.g. dynamics at 400 Hz, IMU at 100 Hz, lidar at 20Hz)?  I think the sensors would have to publish their own LCM types and perhaps run in separate threads for this to be workable.

One potential workaround for the moment would be to run several simulation processes in parallel (one for dynamics, one for laser, etc) at different rates but that seems hacky.

Any thoughts?

liangfok, 2016-03-25:
I vote for running the sensor in a separate process from the simulator and controller, but having a third "clock" process that keeps all simulation / control / sensor processes in sync to the degree necessary.

sherm1, 2016-03-25:
> I vote for running the sensor in a separate process from the simulator and controller

That should be possible, but I would hope not required in general. Could the sensors be treated as discrete blocks with specified update rates? In that case they would fit in nicely within a hybrid simulator without IPC.

liangfok, 2016-03-25:
Yeah, I agree multi-process shouldn't be a requirement. The system should be reconfigurable to be either option. Multi-process provides greater scalability in terms of computational resources but also introduces greater complexity. I also wonder whether multi-process would enable more realistic simulations since it is likely that the real system will be multi-process where separate dedicated computers are used for control vs. perception.

sherm1, 2016-03-25:
> I also wonder whether multi-process would enable more realistic simulations 

Definitely, simulation of communications should be part of an overall simulation. But that wouldn't necessarily require structuring the simulation to mimic the actual structure on the car. For example, if we are doing high-accuracy, non-realtime simulation of the dynamics with ray traced visuals say, then time must flow slowly or at a variable rate for every component of the simulation. Using actual IPC to represent the real IPC could cause trouble, because it would run at full speed. It *could* be better to emulate the communications in order to control time better, maybe as a delay element.

liangfok, 2016-03-25:
Yeah, I agree we shouldn't use actual IPC to represent IPC within a simulation. I was more thinking along the lines of "if the code needs to run as separate processes in a car, might as well allow them to run as separate processes in the simulator to reduce the amount of code changes necessary between sim and reality." The simulation should definitely emulate IPC based on the simulation's real-time factor.

RussTedrake, 2016-05-25:
@psiorx -- perhaps we can close this in favor of any specific new requests?

psiorx, 2016-05-25:
Agreed.  This will come back soon in the context of simulated RGBD sensors when we bring Pete's instantaneous depth map planner into Drake.

closed 2016-05-25 by psiorx