RobotLocomotion/drake #1733: Add c++ support for sending / receiving ROS messages

RussTedrake, 2016-02-21:
Had a good conversation with Brian Gerkey about this, and was asking about how hard it is to act as a ROS node on a machine that wasn’t running the entire ROS ecosystem.  Fantastically he followed up with this:

After we discussed this topic over dinner on Wednesday, I put together
some examples of what I think that you're looking for:
https://github.com/gerkey/ros1_external_use#overview

I didn't change anything in ROS and there's really nothing new here.
I'm just demonstrating a use case that's always been possible but that
we've never paid nearly enough attention to: how to *use* ROS software
without feeling like you're *developing* ROS software. After you've
installed the parts of ROS that you need, you don't have to write
package.xml files, call catkin_make, use roscd, or any of those
things. We find them very useful in our own development, but I
understand why others are not eager to rely on them.

So far I did two use cases, showing how to use vanilla CMake and even
just plain make to build your code on Ubuntu and OSX in a way that
pulls in just what you need from ROS:

* using ROS messages in C++ and Python, including locally defined
custom messages (which requires invoking the code generators), with no
middleware (e.g., if you just want to manipulate the data structures
in-process):
https://github.com/gerkey/ros1_external_use/tree/master/ros1_msgs#overview

* publishing and subscribing ROS messages from C++:
https://github.com/gerkey/ros1_external_use/tree/master/ros1_comms#overview

I'm guessing that your requirements lie in some combination of those
two. If there's something missing, let me know and we can work on
other examples.

The step to install ROS on OSX is a bit lengthy because we don't
produce binary packages for that platform. If it's a priority to
streamline that process, we can look at more automation (e.g., via
brew) and/or producing binary installers (e.g., .dmg files).

While I think that I captured the necessary steps in the instructions,
I did not verify them on a clean VM. If you find anything missing, let
me know (or, better, just submit a pull request!).

psiorx, 2016-02-21:
That's awesome.

It looks similar to what I had to do to get our ROS code building in CLion
and should give us a good starting point for ROS integration.

psiorx, 2016-02-23:
For reference, here's a list of ROS message types that are most relevant to the FLA simulation at this point:

IMU                         sensor_msgs/Imu
2D Lidar                  sensor_msgs/LaserScan
Single Point Lidar    sensor_msgs/Range
pose                        geometry_msgs/PoseStamped
velocity                    geometry_msgs/TwistStamped

We still need to decide on what the actuator command messages should look like.  At their simplest, these could just be `drake_signal_t` lcm types.

RussTedrake, 2016-02-24:
more from @gerkey:

FYI, we've add some more examples to that repo and have started
writing some post-hoc documentation of what's happening:

https://github.com/gerkey/ros1_external_use#preliminary-documentation

gerkey, 2016-02-24:
As you try things out, please let us know how it works (or doesn't) for you.

liangfok, 2016-02-24:
@gerkey, thanks for these tutorials. I successfully ran every one of them except for ros1_actions, which is missing detailed instructions.

gerkey, 2016-02-24:
@liangfok Happy to hear it. I just added a few notes to the actions example (https://github.com/gerkey/ros1_external_use/blob/master/ros1_actions/README.md) that will hopefully get you going. At some point, I'll go back through all the examples to simplify and deduplicate instructions.

liangfok, 2016-02-25:
@gerkey I am wondering what is the best way for Drake to conditionally support ROS based on whether ROS is installed and [left a question on ROS answers](http://answers.ros.org/question/227611/what-is-the-best-way-for-cmakeliststxt-to-check-if-ros-is-installed/). I would appreciate your recommendation.

gerkey, 2016-02-25:
You've got a couple of answers there now.

liangfok, 2016-04-18:
https://github.com/nrjl/lcm_to_ros implements a LCM-to-ROS bridge. Employing a bridge is an alternative to adding native ROS support in Drake. Let's keep it in mind.

RussTedrake, 2016-04-18:
we have many internal versions of this, too.  But we want native ROS support.

liangfok, 2016-05-02:
I'm going to close this issue since it is too broad. More specific Drake / ROS integration issues are:

- https://github.com/RobotLocomotion/drake/issues/2211
- https://github.com/RobotLocomotion/drake/issues/459

There is also PRs https://github.com/RobotLocomotion/drake/pull/2103 and https://github.com/RobotLocomotion/drake/pull/2126 and that configure Drake to be a user of ROS.

closed 2016-05-02 by liangfok