RobotLocomotion/drake #1735: Rolling on flat ground without using Bullet

zamzami, 2016-02-22:
I'm trying to setup an optimisation problem that includes a rolling wheel. Since there is no direct support for bullet collision detection in the optimisation module (also for the known stability reasons), I tried to  hack `getTerrainContactPoints` to add contact for non-zero radius sphere but both implemented methods failed. 

**One contact point on the surface of the sphere**
This solution was initially suggested by posa. the logic behind it is that for a rolling 2D circle on a flat ground the contact point is always the projection of the center on the ground. 
So after adding the following line `pts = getPoints(obj)- [0;0;obj.radius]`
I get the following [planar](https://www.youtube.com/watch?v=FcQ80pc0XDg) & [3D](https://www.youtube.com/watch?v=I6gjwDxRfJI) behaviour. 
The sphere behaves as a pendulum. This may be due to the fact that you are doing some caching on the contact points and not calling `getTerrainContactPoints` each time step

**N contact points distributed over the perimeter of the circle**
So in this method I tried to predefine N points and thought that the LCP will activate/deactivate the appropriate contact point accordingly. 
```matlab        
Npts=128;
         phi=linspace(0,2*pi,Npts);
         x=obj.radius*cos(phi);
         z=obj.radius*sin(phi);
         for i=1:length(x)
        pts(:,i) = getPoints(obj) - [x(i);0;z(i)];
         end 
```
But what actually happened is that two contact constraints get activated in the same time and constraints the body movement.
Here is the video for [2D](https://www.youtube.com/watch?v=oKtpconyW88) & [3D](https://www.youtube.com/watch?v=Clbc_l8uSME) behaviour 

_is there another workaround to achieve rolling behaviour?_ 

PS: The resulting trajectories in all shown videos were achieved using `TimeSteppingRigidBodyManipulator` cascaded with a `foh` input torque for the driving joint

rdeits, 2016-03-09:
`getTerrainContactPoints` should return the terrain contacts in *body frame*. As the wheel rotates, its body frame rotates, so your contact point will be fixed at a single point on the body, which causes the pendulum-like behavior you observed. Unfortunately, I'm not actually sure what the right way to work around this is, since the contact point in body frame will have to depend on the current configuration. But perhaps this will help anyway.

zamzami, 2016-03-10:
Thank you very much for your reply @rdeits.   

I understand that I have a variable contact state problem. But I don't get why it gets stuck with multi-contact points. I thought that by predefining N possible contact points on the surface of the wheel and then letting the complementarity conditions in the optimization handle the activation and deactivation of appropriate contact point it will be possible to achieve a quasi rolling behaviour. 
The stucking behaviour is probably related to the discretization of surface which could leads to conflicting contact constraints if two contacts are activated in the same time. (as mentioned in [ODE FAQ](http://ode-wiki.org/wiki/index.php?title=FAQ#My_rolling_bodies_.28e.g._wheels.29_sometimes_get_stuck_between_geoms)). 

So the other option is to rely on bullet, but bullet doesn't provide smooth gradients and then SNOPT gets nervous with non-smooth gradients, hence the dilemma.

RussTedrake, 2016-05-08:
we are doing a dramatic cleanup of the collision detection <-> optimization solver interface.  our current bullet interface can work with the optimization problems, but the gradients we return ignore the local curvature of the surface so probably won't work well for this.  adding @sherm1 and @amcastro-tri so that this is on their radar.

tri-ltyyu, 2019-03-22:
This is specific to RigidBodyPlant. If this is applicable to MultBodyPlant/SceneGraph, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu