RobotLocomotion/drake #1741: Code base should follow style guide

david-german-tri, 2016-02-23:
We should flesh out the style guide (https://github.com/RobotLocomotion/drake/wiki/Code-Style-Guide), or just adopt the ROS style guide by reference, and then ensure that style is consistent across the code base.  There is a ROS clang-format style that will fix the formatting, but other issues (e.g. trailing underscores for class members) will require manual cleanup.

https://github.com/davetcoleman/roscpp_code_format

jamiesnape, 2016-02-23:
I would recommend adopting either the ROS or Google code styles without modification. Then you have many tools automatically available to you (for instance, if you adopt Google you can use cpplint and include-what-you-use).

billhoffman, 2016-02-24:
@bradking do you have any comments on this?

bradking, 2016-02-24:
I like the use of clang-format.  I have no strong preference on the style itself.

jhoare, 2016-02-24:
I would vote for ROS code guidelines only because I imagine there would be a lot of overlap between ROS users and drake users so not needing to switch styles would be helpful.

psiorx, 2016-02-24:
I'd vote for the Google Style Guide with the exception of file extension naming (I prefer `.cpp` instead of `.cc`).  It's also a lot closer to the existing code we have and is more comprehensive than the ROS counterpart.

Also, the ROS Style guide encourages a number of things that we generally don't like such as passing raw pointers and putting opening curly braces on their own line.

tkoolen, 2016-02-24:
I'm fine with either ROS or Google, with a preference for Google. I agree with John's comments. Additionally, many C++ IDEs (including mine) and text editors come with a built in google style format that you can just switch to; people wouldn't have to create their own.

bradking, 2016-02-24:
Another advantage of Google style is that the `.clang-format` file would be very short:

    ---
    BasedOnStyle:  Google

david-german-tri, 2016-02-24:
I'd prefer the Google style guide for comprehensiveness, and because I trust the team that maintains it. </xoogler-vote>

liangfok, 2016-02-24:
I vote for starting with the Google style guide since its comprehensiveness will likely result in greater consistency. Also, I believe it will be easier to start with stricter guidelines and then to gradually relax them rather than the other way around.

david-german-tri, 2016-02-25:
There seems to be a pretty strong majority for the Google style guide.  Let's start following it in new pull requests.  I'll add a .clang-format today.  Of course, there are plenty of style violations that the formatter can't handle, so those will have to be fixed up manually over time as we touch the code.  [cpplint](https://pypi.python.org/pypi/cpplint) will be useful.

Here at TRI yesterday, we discussed a multi-level code review process to take some load off of @RussTedrake.  The proposal is:

1. Get a preliminary code review from someone else.  I think ideally this would be cross-team (Kitware reviews TRI, TRI reviews Kitware) to spread knowledge.

2. Get an approval code review from @RussTedrake.  Ideally, there will be few issues requiring comment from Russ, and as a result he'll be able to approve more pull requests faster.

Any objections to giving this a try?  In the Kitware->TRI direction, you can load-balance across myself and @liangfok.  Who should we ping on the Kitware side?

tkoolen, 2016-02-25:
I would be strongly in favor of first having a single pull request that only reformats (using an automated tool) all of the existing code without changing functionality, instead of only enforcing the style in new pull requests. 

If we don't reformat existing code first, diffs in pull requests will involve a lot of formatting changes, making code reviews harder. Plus you can't really do the formatting correctly only in the parts of a file that you've changed (e.g. indentation will be annoyingly inconsistent between parts of the file that you've changed and untouched parts). Not reformatting first would also make this transition drag on forever.

david-german-tri, 2016-02-25:
Good point, how about this:

1. I'll reformat the entire code base in one shot using clang-format today.
2. Style requirements not related to formatting will be enforced on new pull requests in code review.

rdeits, 2016-02-25:
Exciting :). And I agree with @tkoolen. Having accidentally caused massive diffs by having "eliminate trailing whitespace" turned on in my editor, I'm all for starting from a consistent style.

jamiesnape, 2016-02-25:
I can set up jobs on Jenkins that run cpplint and [include-what-you-use](http://include-what-you-use.org/). They will cause a lot of noise at first, so it may be best to have them just run nightly or on-demand for the first few days.

jamiesnape, 2016-02-25:
If anyone would like other tools to run on Jenkins, create a ticket and ping me on it.

RussTedrake, 2016-02-26:
it seems like once negative consequence of this idea is that @david-german-tri will suddenly be the last committer for basically every line in the cpp files (git blame will always blame you).  Is that an acceptable loss?

david-german-tri, 2016-02-26:
I'm inclined to think it is. Even if we reformatted incrementally, we'd
have the same problem on a file by file basis. It's easy enough to adjust
the git-blame revision to look past.

RussTedrake, 2016-02-26:
OK. Agreed.

iamwolf, 2016-02-26:
@david-german-tri Out of curiosity: On a local machine ``git blame -w -M`` ignores formatting changes (apart from linebreaks), how is it handled on [GitHub](https://help.github.com/articles/using-git-blame-to-trace-changes-in-a-file/)?

bradking, 2016-02-26:
> it seems like once negative consequence of this idea is that david-german-tri will suddenly be the last committer for basically every line in the cpp files 

For automated conversions like that I typically use a "robot" user to take the blame.  That makes it clear that something automated was done when such lines are seen in the blame output.

I also record the script (or equivalently precise instructions) used to do the conversion in the commit message.  That way anyone can reproduce the conversion, and anyone with open topics can run the conversion themselves to help them rebase cleanly.

RussTedrake, 2016-02-26:
Great suggestions.

[https://github.com/RobotLocomotion/drake/issues/1741#issuecomment-189271507] .[https://github.com/notifications/beacon/AGJNNFh9kk2VzlJCWcaRx5TeucinqxXCks5poEoBgaJpZM4Hg9Lw.gif]

david-german-tri, 2016-02-26:
@iamwolf I don't see a way to change the actual blame command on GitHub, but you can move backwards in time on the History tab and the blame updates accordingly.

RussTedrake, 2016-02-28:
just spent some time (for the first time), reading the google style guide more carefully.  there are going to be a lot of painful changes.  (ex: using namespace Eigen, using namespace std, which we have at the top of most of our .cpp files are not allowed; many variable names will change; ... ).  i support this and will comply, but it's going to take time and patience to completely conform.

as much as I prefer .cpp to .cc, it seems pretty arbitrary for us to accept the entire style guide except for this rule.  i know we'll be making some amendments to the style guide, but i don't think this one is valuable enough to warrant non-compliance.  if we're going all in, then we should probably only make exceptions on moral grounds, or because of the particular demands of our codebase, not because of our own coding style inertia.

jamiesnape, 2016-03-01:
@RussTedrake I agree. The initial pain will be worth it in the long term.

jwnimmer-tri, 2016-03-03:
> using namespace Eigen, using namespace std, which we have at the top of most of our .cpp files are not allowed

I would support a variance to the guide to generally allow 'using namespace foo;' statements in *.cc files (but still forbid them in *.h files).  The pollution argument cited by google-styleguide is valid, but I'd rather trust our committers to use good judgement and write clear code -- and sometimes the clearest code will be with 'using namespace ...' statements atop the *.cc file.

jamiesnape, 2016-03-03:
I would say avoiding `using namesphere` anywhere is a good thing, especially `using namespace std`. You can use a namespace alias for code clarity.

david-german-tri, 2016-03-03:
I agree with @jamiesnape.  When namespaces are verbose, the style guide permits `using really_long_namespace::Bar;` statements for the specific names that a given .cc file uses.  In addition to keeping pollution under control and reducing collision risk (versus `using` the whole namespace), this approach documents where the name came from.

liangfok, 2016-03-03:
I vote for avoiding the use of "using namespace foo" since it requires someone to guess where certain names come from. I would rather have numerous `using namespace::Bar` statements at the top of the file to avoid that confusion.

jwnimmer-tri, 2016-03-03:
Sounds like a consensus, then, and OK with me.

david-german-tri, 2016-03-24:
The Google Style Guide has been officially adopted in the docs:

https://github.com/RobotLocomotion/drake/blob/277817621bfc7a381755a109f7378253b4118978/drake/doc/code_style_guide.rst#L14

closed 2016-03-24 by david-german-tri