RobotLocomotion/drake #1745: suggested updates to the install process on Ubuntu ...

mjhmilla, 2016-02-24:
Installing Drake on Ubuntu was pretty smooth for such a large and complex code base - well done! Along the way I encountered a few things that you might update/change to make it even easier the next time around.

**I) Installation:** I had to update the following files to get Drake to fully install in ubuntu:
1. drake_ubuntu/install_prereqs.sh:
    changed line 24 from `SUBDIRS="drake externals` to `SUBDIRS="externals" `
2. drake_ubuntu/externals/install_prereqs.sh
    changed 2 occurances of : `../pod-build/drake_external_source_dirs.txt` to `../drake_external_source_dirs.txt`

**II) Install documentation:** It would also be helpful to add 2 notes to the documentation:
1. When running cmake leave the CMAKE_INSTALL_PREFIX set to 'drake_distro/build'. Why? I changed this folder and then found I had to update addpath_drake.m to work ... after which I decided to stick with the original folder name to prevent further problems. 
2. Under 'Build the collection' add a step:
`cd drake_distro/drake`
 `make`
Why? Otherwise the rimless wheel and monopod hopper examples don't run because the mex files haven't been compiled. At the moment when one of these examples fails it actually tells you to complete the above step, which is pretty thoughtful.
    
**III) Remaining build errors:**  I still have build errors for some of the external libraries, though they are not show-stoppers. As a note, in the cmake file I set `WITH_ALL_PUBLIC_EXTERNALS` to `ON`. Yes, I've run make multiple times and this one doesn't go away. Here's the error:

`make[9]: *** [src/CMakeFiles/iris_geometry.dir/geometry.cpp.o] Error 1`
`make[8]: *** [src/CMakeFiles/iris_geometry.dir/all] Error 2`
`make[7]: *** [all] Error 2`
`make[6]: *** [iris_project-prefix/src/iris_project-stamp/iris_project-build] Error 2`
`make[5]: *** [CMakeFiles/iris_project.dir/all] Error 2`
`make[4]: *** [all] Error 2`

RussTedrake, 2016-02-24:
Very helpful, thanks. We just recently updated IRIS. If you can show us 
more of
the error output, that would help us debug.

mjhmilla, 2016-02-24:
Attached is all of the command prompt output from calling make in
drake_distro. I also went into externals/iris and ran make - that output is
attached as well. That output is much more useful:

----------
In file included from
/home/mjhmilla/dev/drake/externals/iris/iris/src/geometry.cpp:5:0:
/home/mjhmilla/dev/drake/externals/iris/iris/src/iris_cdd.cpp:2:21: fatal
error: setoper.h: No such file or directory
 #include "setoper.h"
----------

The file "setoper.h" is in

/home/mjhmilla/dev/drake/externals/iris/pod-build/cdd_project-prefix/src/cdd_project/lib-src
and a link to it
/home/mjhmilla/dev/drake/externals/iris/pod-build/cdd_project-prefix/src/cdd_project/lib-src-gmp

so I think a fix is at hand with an edit to IRIS's Cmake file. Is it a
problem that my root directory is named 'drake' rather than 'drake_ubuntu'
or something like that?

Matt

-- CMAKE_INSTALL_PREFIX=/home/mjhmilla/dev/drake/build
-- Preparing to build cmake with dependencies: 
-- Preparing to build textbook with dependencies: 
-- Preparing to build eigen with dependencies: 
-- Preparing to build lcm with dependencies: 
-- Preparing to build libbot with dependencies: lcm
-- Preparing to build bullet with dependencies: 
-- Preparing to build spotless with dependencies: 
-- Preparing to build director with dependencies: lcm;libbot
-- Preparing to build signalscope with dependencies: director
-- Preparing to build octomap with dependencies: 
-- Preparing to build mosek with dependencies: 
-- Preparing to build iris with dependencies: eigen;mosek
-- Preparing to build yalmip with dependencies: 
-- Preparing to build avl with dependencies: 
-- Preparing to build xfoil with dependencies: 
-- Preparing to build meshconverters with dependencies: 
-- Preparing to build swigmake with dependencies: 
-- Preparing to build swig_matlab with dependencies: 
-- Preparing to build drake with dependencies: cmake;eigen;lcm;libbot;bullet;octomap;swig_matlab;swigmake
-- Installation will include extra example: LittleDog
-- Preparing to install precompiled snopt
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mjhmilla/dev/drake
[  0%] [  1%] [  1%] [  2%] [  2%] [  2%] Performing update step for 'download-avl'
Performing update step for 'download-cmake'
[  2%] Performing update step for 'download-director'
Performing update step for 'download-libbot'
Performing update step for 'download-lcm'
Performing update step for 'download-eigen'
Performing update step for 'download-bullet'
[  2%] Performing update step for 'download-LittleDog'
[  2%] [  2%] [  2%] [  2%] [  2%] [  3%] [  4%] No configure step for 'download-libbot'
No configure step for 'download-bullet'
No configure step for 'download-lcm'
No configure step for 'download-avl'
No configure step for 'download-director'
No configure step for 'download-cmake'
No configure step for 'download-eigen'
[  4%] [  4%] [  4%] [  5%] [  6%] No configure step for 'download-LittleDog'
[  6%] [  6%] No build step for 'download-avl'
No build step for 'download-bullet'
[  6%] No build step for 'download-libbot'
No build step for 'download-director'
No build step for 'download-lcm'
[... 714 lines cut ...]
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/NonLinearOptimization/lmpar.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/NonLinearOptimization/rwupdt.h
-- Installing: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/CMakeLists.txt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineMatrix.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineProduct.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineInplaceLU.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineMatrixBase.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineUtil.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/Skyline/SkylineStorage.h
-- Installing: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/BVH
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/BVH/CMakeLists.txt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/BVH/BVAlgorithms.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/BVH/KdBVH.h
-- Installing: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/CMakeLists.txt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/CopyrightMINPACK.txt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/LevenbergMarquardt.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/LMpar.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/LMcovar.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/LMqrsolv.h
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/LevenbergMarquardt/LMonestep.h
-- Installing: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/MatrixFunctions
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/MatrixFunctions/CMakeLists.txt
-- Up-to-date: /home/mjhmilla/dev/drake/build/include/eigen3/unsupported/Eigen/src/MatrixFunctions/MatrixFuncti

rdeits, 2016-02-24:
I think something has gone wrong with your build process. The fact that you had to change `../pod-build/drake_external_source_dirs.txt` to `../drake_external_source_dirs.txt` probably indicates that cmake is using the wrong folder for its build byproducts (it sounds like it's putting them directly into `drake_ubuntu` instead of `drake_ubuntu/pod-build`, and that's likely also why you had to do an additional `make` inside `drake`, and possibly why you're having problems with `IRIS`. We test the ubuntu installation process automatically through Travis (including IRIS), so I'm reasonably confident that it should work without modification. 

So maybe the best thing to do is to go back to the beginning and try to find the root cause. Why did you have to remove `drake` from the `install_prereqs.sh` file? 

And to fix your build, I think that a fresh clone of drake is likely to be the easiest thing to do. All you should need to do from that is `cd` into the cloned repo and do:

```bash
make options # and enable any extra packages you want to install
make download-all
sudo ./install_prereqs.sh ubuntu
make
```

mjhmilla, 2016-02-24:
Thank you for the help!

I started with a fresh repo with the root folder name 'drake_ubuntu'.
Everything is fine until 'sudo ./install_prereqs.sh ubuntu' which has the
output

--------------------------------------------------
...build-essential is already the newest version.
cmake is already the newest version.
swig is already the newest version.
openjdk-6-jdk is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
install_prereqs could not find the list of *external source directories*.
Please run 'make download-all' first.
--------------------------------------------------

When I go and look at 'install_prereqs.sh' I see on line 23

--------------------------------------------------
*SUBDIRS="drake externals"*
for subdir in $SUBDIRS; do
  if [ -f $subdir/install_prereqs.sh ]; then
...
--------------------------------------------------

but I don't see a directory called 'drake externals' in 'drake_ubuntu'. I
only see 'externals'. Inside of 'externals' I see what looks like the
'$subdir/install_prereqs.sh' that the script is trying to call. That's why
I changed "drake externals" to "externals" earlier. When I did this earlier
I ended up with everything working (as far as the examples are concerned)
except for iris.

What should have I done?

Thanks again,

Matt

mjhmilla, 2016-02-24:
The error message I was seeing was coming from externals/install_prereqs.sh
when this if statement fails:

----------------------------------------
if [ -e "../pod-build/drake_external_source_dirs.txt" ]
then
  SUBDIRS=`cat ../pod-build/drake_external_source_dirs.txt`
else
  echo "install_prereqs could not find the list of external source
directories. Please run 'make download-all' first." 1>&2
  exit 1
fi
----------------------------------------

and its failing because I don't have a folder called 'pod-build' in
'drake_ubuntu'.

Any idea why there's no 'pod-build' directory in 'drake-ubuntu'?

Thanks again!

Matt

rdeits, 2016-02-24:
Did you run `make download-all`?

mjhmilla, 2016-02-24:
Yes

mjhmilla, 2016-02-24:
The command prompt output from running 'make download all' is attached, in
case it helps. I see no mention of 'pod', or 'pod-build'.

Thanks again for the help - I really do appreciate it.

Matt

-- CMAKE_INSTALL_PREFIX=/home/mjhmilla/dev/drake_ubuntu/build
-- Preparing to build cmake with dependencies: 
-- Preparing to build textbook with dependencies: 
-- Preparing to build eigen with dependencies: 
-- Preparing to build lcm with dependencies: 
-- Preparing to build libbot with dependencies: lcm
-- Preparing to build bullet with dependencies: 
-- Preparing to build spotless with dependencies: 
-- Preparing to build director with dependencies: lcm;libbot
-- Preparing to build signalscope with dependencies: director
-- Preparing to build octomap with dependencies: 
-- Preparing to build mosek with dependencies: 
-- Preparing to build iris with dependencies: eigen;mosek
-- Preparing to build yalmip with dependencies: 
-- Preparing to build avl with dependencies: 
-- Preparing to build xfoil with dependencies: 
-- Preparing to build meshconverters with dependencies: 
-- Preparing to build swigmake with dependencies: 
-- Preparing to build swig_matlab with dependencies: 
-- Preparing to build drake with dependencies: cmake;eigen;lcm;libbot;bullet;octomap;swig_matlab;swigmake
-- Installation will include extra example: LittleDog
-- Preparing to install precompiled snopt
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mjhmilla/dev/drake_ubuntu
[  4%] Built target download-snopt-precompiled
[  4%] Performing update step for 'download-cmake'
[  6%] No configure step for 'download-cmake'
[  6%] No build step for 'download-cmake'
[  6%] No install step for 'download-cmake'
[  6%] Completed 'download-cmake'
[  8%] Built target download-cmake
[  8%] Performing update step for 'download-textbook'
Your branch is up-to-date with 'origin/master'.
[ 10%] No configure step for 'download-textbook'
[ 10%] No build step for 'download-textbook'
[ 10%] No install step for 'download-textbook'
[ 12%] Completed 'download-textbook'
[ 14%] Built target download-textbook
[ 14%] Performing update step for 'download-libbot'
[ 14%] No configure step for 'download-libbot'
[ 14%] No build step for 'download-libbot'
[ 14%] No install step for 'download-libbot'
[ 14%] Completed 'download-libbot'
[ 18%] Built target download-libbot
[ 18%] Performing update step for 'download-eigen'
[ 20%] No configure step for 'download-eigen'
[ 20%] No build step for 'download-eigen'
[ 20%] No install step for 'download-eigen'
[ 22%] Completed 'download-eigen'
[ 24%] Built target download-eigen
[ 24%] Performing update step for 'download-octomap'
[ 24%] No configure step for 'download-octomap'
[ 24%] No build step for 'download-octomap'
[ 26%] No install step for 'download-octomap'
[ 26%] Completed 'download-octomap'
[ 28%] Built target download-octomap
[ 30%] Performing update step for 'download-bullet'
[ 30%] No configure step for 'download-bullet'
[ 32%] No build step for 'download-bullet'
[ 32%] No install step for 'download-bullet'
[ 34%] Completed 'download-bullet'
[ 34%] Built target download-bullet
[... 54 lines cut ...]
[ 81%] Performing update step for 'download-xfoil'
[ 81%] No configure step for 'download-xfoil'
[ 81%] No build step for 'download-xfoil'
[ 81%] No install step for 'download-xfoil'
[ 81%] Completed 'download-xfoil'
[ 85%] Built target download-xfoil
[ 87%] Performing update step for 'download-swigmake'
[ 87%] No configure step for 'download-swigmake'
[ 87%] No build step for 'download-swigmake'
[ 89%] No install step for 'download-swigmake'
[ 89%] Completed 'download-swigmake'
[ 89%] Built target download-swigmake
[ 89%] Performing update step for 'download-swig_matlab'
[ 91%] No configure step for 'download-swig_matlab'
[ 91%] No build step for 'download-swig_matlab'
[ 91%] No install step for 'download-swig_matlab'
[ 91%] Completed 'download-swig_matlab'
[ 93%] Built target download-swig_matlab
[ 93%] Performing update step for 'download-LittleDog'
[ 93%] No configure step for 'download-LittleDog'
[ 95%] No build step for 'download-LittleDog'
[ 95%] No install step for 'download-LittleDog'
[ 97%] Completed 'download-LittleDog'
[100%] Built target download-LittleDog
[100%] Built target download-all

rdeits, 2016-02-24:
Ah, okay. This looks like the output of `make download-all` after it's already been run at least once. So whatever strange mis-configuration of the build system is still present. It will help me a lot if you can try again from a clean clone of drake. Can you please try the following?

```
git clone https://github.com/RobotLocomotion/drake.git drake-clean
cd drake-clean
mkdir build
make options
make download-all
```

and see if that results in the `pod-build` folder being created? 

If it does, then you should be able to run `./install_prereqs.sh ubuntu` and have it run without errors.

mjhmilla, 2016-02-24:
Problem found, and I'm it. When I saw this instruction:

make options # use the gui to choose which externals you ...

I thought 'don't they mean ccmake?'. And so I ran ccmake and this skipped
script in 'make options' you've written to make the directory pod-build.
I've never run 'make options' before and that's why I switched over to
ccmake without thinking about it. Three times. Everything is installing
right now, and I expect it will work just fine.

Thank you for your patience,

Matt

rdeits, 2016-02-24:
Okay, that makes sense. I'm glad you figured it out!

closed 2016-02-25 by RussTedrake